WsprryPi/WsprryPi #51: Pi Zero W issue unattended boot

edegraaff, 2024-02-10:
Hi, 

A great project, and i want to built several wspr nodes for some other HAM's here in NL. I have one issue with booting, the Pi Zero W, it seems the node only boots when i attach the monitor, the usb keyboard is not needed at all. I have no clue why this hapening, i have a lot of pi's running in unattended mode with no issues. Does someone recognise this issue, and have suggestions to solve? Could it have relation with the activation of the wspr transmition?

73 de pd1eg
Eelco

lbussy, 2024-02-21:
Unable to reproduce - are you still having this issue?

edegraaff, 2024-02-21:
> Unable to reproduce - are you still having this issue?

Hi Lee, also for others, since i could not reproduce on the 2 new pi zero w models, i asked myself what is the difference, mine is from 2021. Asked arroud nobody knew the issue. My board is a v1.1 the recent ones a v 1.2 if i remember right. Well lets try something that is not adviced but... sudo apt full-upgrade did not solve and then i did as root rpi-update and this solved the problem until now.

lbussy, 2024-02-25:
After several emails - this seems to clearly be a Raspberry Pi / Raspbian issue.  This only impacts the OG Raspberry Pi Zero (without WiFi.)

Closing as upstream.

closed 2024-02-25 by lbussy

edegraaff, 2024-02-25:
> After several emails - this seems to clearly be a Raspberry Pi / Raspbian issue. This only impacts the OG Raspberry Pi Zero (without WiFi.)
> 
> Closing as upstream.

Agree, i think it is PI Zero W 1.1  related, the other new rev 1.2 devices for other HAM's work fine.

trancen, 2024-05-03:
I ran across the same issue with the older RPiZw(v1.1 I had a good 6-8 because I was getting them at $5 each), After spending what seems like a 100 hrs trying to figure it out. The solution to get around buying a newer pi was to use whats called a dummy HDMI dongle(about $3). I can confirm that after getting the dummy hdmi this fork of WspryPi (and others) works headless now.

lbussy, 2024-05-05:
I cannot find any information about this anywhere.  Are you sure you are using the light version of the OS and SSH connections?

trancen, 2024-05-06:
I have tried a few versions of Raspberry Pi OS (Legacy) Lite and none Legacy as well down 2019-2024

It took me a while to find the issue as well after  I spent a LOT of time trying to figure it out myself. The link below is where it started to resolve my issues. https://github.com/JamesP6000/WsprryPi/issues/6

edegraaff, 2024-05-06:
hi Lee, i am now testing tje pi zero w with the dongle keep ypu posted the next few days...

⁣Follow us @ https://www.fishfreak.org Fly-fish & -tie, Nordics, camp & outdoor​

edegraaff, 2024-05-07:
hi Lee, witj tje dongle the spi zero still operational, it seems a solution for the systen "hang" issue. I think the hf signal created by pen no 4 could disturb one way or another someting on the li board. The good nes is it seems the dongle does solve the issue with unattended pi's that crash after a while when they operating withoud an active monitor attached.

regards eelco de pd1eg

⁣Follow us @ https://www.fishfreak.org Fly-fish & -tie, Nordics, camp & outdoor​

trancen, 2024-05-07:
I am as well Lee, this is basically the setup I'm using . 

[WSPR HAT](https://imgur.com/a/UkGsvpM)

edegraaff, 2024-05-07:
Ok, i whill share a picture of my configuration. The good news is yes the dongle works, so issue solved for 2.5 euro. I has costed us months to figure out a solution, this one is so simple, never know there are dongles for this issue... 

[image]

edegraaff, 2024-05-17:
Foor all to know: My pi zero W 2017, V1.1 did work for a few days perfectly with the HDMI dongle, and then .. it seemed dead.. after rebooting, the print became red-hot, disconnecting the dongle did not help, the system did boot, but when the os was in final state the zero dropped to poweroff mode, and was again hot! It could be a dongle on the pi zero does make some sort of resistance that drives the HDMI port to hot... anyway:  in the junk: new pi zero 2w and the system is running without a dongle, headless..  :-)