WsprryPi/WsprryPi #56: New Install - wspr executable error

AA7US, 2024-02-20:
Greetings,

I just installed Wsprry Pi 1.2.0 on a trusty old Pi 3 using a fresh download copy of 32 bit Raspberry Pi OS Legacy (Bullseye).  The install of Wsprry Pi went fine.

After configuring the web GUI, WSPR beacons do not transmit. 

Checking the wspr daemon status from the command line I get this (**BOLD** added for emphasis):

pi@WSPRPi:~ $ sudo systemctl status wspr
● wspr.service - wspr daemon for: wspr
     Loaded: loaded (/etc/systemd/system/wspr.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Mon 2024-02-19 19:00:49 MST; 4s ago
       Docs: https://github.com/lbussy/WsprryPi/discussions
    Process: 854 ExecStart=/usr/local/bin/wspr -D -i /usr/local/etc/wspr.ini (**code=exited, status=1/FAILURE**)
   Main PID: 854 (**code=exited, status=1/FAILURE**)
        CPU: 6ms

When I run "sudo wspr -h" I get this:

NOTE: I removed some single quotation marks to preserve formatting here.

pi@WSPRPi:/usr/local/bin $ sudo wspr -h
wspr: /lib/arm-linux-gnueabihf/libc.so.6: version GLIBC_2.32 not found (required by wspr)
wspr: /lib/arm-linux-gnueabihf/libc.so.6: version GLIBC_2.34 not found (required by wspr)
wspr: /lib/arm-linux-gnueabihf/libc.so.6: version GLIBC_2.33 not found (required by wspr)
wspr: /lib/arm-linux-gnueabihf/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by wspr)

Any guidance on how to correct this will be most appreciated.

73,

AA7US

lbussy, 2024-02-21:
Thanks for the submission.  I was indeed notified of this by another user.  I have reverted to version 1.1.0 of the executable, there were no changes to that part, and I am testing now.

lbussy, 2024-02-21:
You can re-run the installer.  The script should prompt you to install an OLDER version of the `wspr` executable.  Do that, and all should be well.

lbussy, 2024-02-21:
Very good, thanks for letting me know!

lbussy, 2024-03-06:
Also see #65