WsprryPi/WsprryPi #57: Request: Beacon Timing via Web GUI

AA7US, 2024-02-21:
Please consider adding the ability to control how often WSPR beacons are transmitted using the web GUI as can be done from the command line.

For example, to transmit one WSPR beacon on the 30M band every 8 minutes (repeating indefinitely), a sample command is:

wspr -r -s N9NNN EM10 20 1014020 0 0 0

It sure would be convenient to accomplish the same thing using the web GUI. :)

73,

AA7US

lbussy, 2024-02-22:
Like this:

[image]

AA7US, 2024-02-22:
Perfect!  I didn't realize this could be entered into the web GUI.  Thank you!

lbussy, 2024-02-22:
Yer welcome ... I'm glad to be abe to close one :)

AA7US, 2024-02-22:
Well I hate to ruin a celebration, but unfortunately the zeros I just tried in the frequency field aren't being honored.  I tried both "30m 0 0 0" and "10140200 0 0 0" and in both cases my Pi continued to transmit during every 2 minute window. :(

lbussy, 2024-02-22:
"Works here?"

I'll have a look.

lbussy, 2024-02-23:
Probably related to #58

lbussy, 2024-02-25:
When I tested, it treated all frequencies in the list as `0` (skip):

```
pi@wspr:~/WsprryPi/src $ sudo ./wspr --repeat --terminate 7 AA0NT EM18 33 20m 0 0 0 0
Wsprry Pi v1.2.0-3-g1fb27ee (freq_list).
Running on: Raspberry Pi 2B or 3B (BCM2837).
WSPR packet payload:
- Callsign: AA0NT
- Locator:  EM18
- Power:    33 dBm
Requested TX frequencies:
- 14.097100 MHz
- 0.000000 MHz
- 0.000000 MHz
- 0.000000 MHz
- 0.000000 MHz

Extra options:
- Using NTP to calibrate transmission frequency.
- TX will stop after 7 transmissions.

Ready to transmit (setup complete).
Center frequency for WSPR trans: 14.097100 MHz.
Waiting for next WSPR transmission window.
Obtained new ppm value: 2.409317
Skipping transmission.
Center frequency for WSPR trans: 0.000000 MHz.
Waiting for next WSPR transmission window.
Obtained new ppm value: -0.571365
Skipping transmission.
Center frequency for WSPR trans: 0.000000 MHz.
Waiting for next WSPR transmission window.
Exiting, caught signal: 2
```

... which is still an issue but not exactly what you had.

AA7US, 2024-02-25:
Ah the nasty ol' interrupt signal.

By chance did you navigate away from the terminal window (say changed focus to another window on your PC) then come back to it?

I often experience this interrupt signal when using the original WsprryPi program when going back to a terminal session.  I avoid this problem by running it (WsprryPi) in a tmux session which I then “detach” away from leaving WsprryPi running.

Even then, whenever I go back to the tmux session (i.e.: “reattach” to it), doing that usually kills WsprryPi with an interrupt signal.

In short, it’s annoying but not new.  It's been complained about over the years in WsprryPi issues.

lbussy, 2024-02-25:
Nah that was me hitting ctrl-c to exit.  The issue there was when it got a non-zero frequency, it skipped transmission.

The first try there should have been 20m but it was skipped - that's the bug as I see it.

lbussy, 2024-03-01:
This is still on my list - work actually expects me to work so I am slower than I want to be.

lbussy, 2024-03-05:
Still scratching my head, interspersed with bricking a few SD cards while testing.

lbussy, 2024-03-06:
Well, it freaking works now.

```
$ sudo ./wspr --terminate 2 AA0NT EM18 33 20m 0 0
Wsprry Pi v1.2.1-Alpha.1-1-g64cf608 (freq_list).
Running on: Raspberry Pi 4 (BCM2711).
WSPR packet payload:
- Callsign: AA0NT
- Locator:  EM18
- Power:    33 dBm
Requested TX frequencies:
- 14.097100 MHz
- 0.000000 MHz
- 0.000000 MHz

Extra options:
- Using NTP to calibrate transmission frequency.
- TX will stop after 2 transmissions.

Ready to transmit (setup complete).
Center frequency for WSPR trans: 14.097100 MHz.
Waiting for next WSPR transmission window.
Obtained new ppm value: -19.485550
Transmission started.
Transmission completed, (110.606530 sec.)
Center frequency for WSPR trans: 0.000000 MHz.
Waiting for next WSPR transmission window.
Skipping transmission.
Center frequency for WSPR trans: 0.000000 MHz.
Waiting for next WSPR transmission window.
Skipping transmission.
```

This is a newly compiled version so I am not sure if a library difference is at play here.

There is a remaining issue where `--repeat` and `--terminate conflict`, possibly including a frequency list.  I need to figure that out - is this an issue or is this desired?  More to come.

I need to test the same from the web UI, then maybe I'll release an Alpha version later today.  The question about `-r` and `-t` will be after that.

lbussy, 2024-03-06:
@AA7US please try this:

`curl -L https://raw.githubusercontent.com/lbussy/WsprryPi/freq_list/scripts/install.sh | sudo bash`
(all one line)

As always, this is idempotent, and you can run it over and over.  This is an Alpha branch and will go away (just in case someone sees this later on.)

Theoretically, this works with a frequency list, with the above caveats.

AA7US, 2024-03-06:
Hi Lee,

Thanks for all your hard work on this.  I just ran the alpha install script, rebooted, but unfortunately the wspr daemon is failing to load again.  Presumably this is the same problem we covered in [Issue #56](https://github.com/lbussy/WsprryPi/issues/56).

John L.

```
pi@WSPRPi:~ $ sudo systemctl status wspr
● wspr.service - wspr daemon for: wspr
     Loaded: loaded (/etc/systemd/system/wspr.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2024-03-06 13:09:10 MST; 2s ago
       Docs: https://github.com/lbussy/WsprryPi/discussions
    Process: 744 ExecStart=/usr/local/bin/wspr -D -i /usr/local/etc/wspr.ini (code=exited, status=1/FAILURE)
   Main PID: 744 (code=exited, status=1/FAILURE)
        CPU: 7ms 
```

lbussy, 2024-03-06:
Have you rebooted?  I added another "known issue" for that.

AA7US, 2024-03-06:
> Have you rebooted? I added another "known issue" for that.

Yes.

lbussy, 2024-03-06:
Can you give me the last few lines of `/var/log/wspr/wspr.error.log` please?

Was this a new install or a clean flash?

AA7US, 2024-03-06:
> Can you give me the last few lines of `/var/log/wspr/wspr.error.log` please?

All of the entries in wspr.error.log are:

```
/usr/local/bin/wspr: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.32' not found (required by /usr/local/bin/wspr)
/usr/local/bin/wspr: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.34' not found (required by /usr/local/bin/wspr)
/usr/local/bin/wspr: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /usr/local/bin/wspr)
/usr/local/bin/wspr: /lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/local/bin/wspr)
```
>
> Was this a new install or a clean flash?

I installed the alpha over existing wsprrypi release v1.1.0, which was working fine except for the tx timing problem.

Full history in case it matters:

Back in February I started with a clean install of Bullseye (Legacy) Lite downloaded from the Raspberry Pi Foundation.  I configured a few things for my network and installed chrony in place of the standard ntp client.  I run OS updates.  I then installed wsprrypi.  Since then, I've only run and rerun your wsprrypi install script.

If you want me to try a new Bullseye install followed by an alpha wsprrypi install, I can certainly do that.

John L.

lbussy, 2024-03-06:
Interesting - let me look at that a bit.  If you hit it, others will.

lbussy, 2024-03-06:
Can you give me the results of `ldd --version` please?  It should be like:

```
ldd (Debian GLIBC 2.36-9+rpt2+deb12u3) 2.36
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper
```

Also `strings $(/sbin/ldconfig -p | grep stdc++|awk -F'=>' '{print $2}')|grep LIBCXX`.  It should look like:

```
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_3.4.26
GLIBCXX_3.4.27
GLIBCXX_3.4.28
GLIBCXX_3.4.29
GLIBCXX_3.4.30
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

AA7US, 2024-03-06:
```
pi@WSPRPi:~ $ ldd --version
ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u7) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

```
pi@WSPRPi:~ $ strings $(/sbin/ldconfig -p | grep stdc++|awk -F'=>' '{print $2}')|grep LIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_3.4.26
GLIBCXX_3.4.27
GLIBCXX_3.4.28
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

Compared to you, it looks like I'm missing some versions?

lbussy, 2024-03-06:
Okay please see if this addresses it.  If so I can add these to the install script:

`sudo apt-get install libstdc++6 libc6`

AA7US, 2024-03-06:
Unfortunately, no joy. 🙁

```
pi@WSPRPi:~ $ sudo apt-get install libstdc++6 libc6
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libstdc++6 is already the newest version (10.2.1-6+rpi1).
libc6 is already the newest version (2.31-13+rpt2+rpi1+deb11u7).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

AA7US, 2024-03-06:
If it helps any...

```
pi@WSPRPi:~ $ sudo cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

lbussy, 2024-03-06:
That's just ... weird.  It says it has them and doesn't show the latest.  

`sudo apt update' then try the install again?

AA7US, 2024-03-06:
```
pi@WSPRPi:~ $ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

pi@WSPRPi:~ $ sudo apt-get install libstdc++6 libc6
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libstdc++6 is already the newest version (10.2.1-6+rpi1).
libc6 is already the newest version (2.31-13+rpt2+rpi1+deb11u7).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

AA7US, 2024-03-06:
> That's just ... weird. It says it has them and doesn't show the latest.

Strange indeed. 🤔

lbussy, 2024-03-06:
I think I can go back to the image you have and try as well.

lbussy, 2024-03-06:
One last thing to try:
```
sudo apt update
sudo apt-get install libc-bin libc6 locales
sudo apt-get install libstdc++6 libc6 libc-bin libc6 locales
```

lbussy, 2024-03-06:
Grr!

Well, I know what's needed, not how to get it (yet).

lbussy, 2024-03-07:
Okay, wanna give it another go?  I statically linked the compiler libs.  If this doesn't work, I need to try a different path.

AA7US, 2024-03-07:
Good morning Lee,

I reran the alpha script.  Unfortunately the problem is still there.

Below are the entries from install.log followed by a wspr daemon status (failure exit).  There are GLIBC not found entries in the install log now that didn't appear during previous installs.

John L.

```
2024-03-07 06:25:59 ***Script install.sh starting.***
2024-03-07 06:25:59 Running on: Raspbian GNU/Linux 11 (bullseye)
2024-03-07 06:25:59 You will be presented with some choices during the install. Most frequently
2024-03-07 06:25:59 you will see a 'yes or no' choice, with the default choice capitalized like
2024-03-07 06:25:59 so: [y/N]. Default means if you hit <enter> without typing anything, you will
2024-03-07 06:25:59 make the capitalized choice, i.e. hitting <enter> when you see [Y/n] will
2024-03-07 06:26:00 default to 'yes.'
2024-03-07 06:26:00 Yes/no choices are not case sensitive. However; passwords, system names and
2024-03-07 06:26:00 install paths are. Be aware of this. There is generally no difference between
2024-03-07 06:26:00 'y', 'yes', 'YES', 'Yes'.
2024-03-07 06:26:01 Press any key when you are ready to proceed.
2024-03-07 06:26:01 The time is currently set to Thu 7 Mar 06:26:01 MST 2024.
2024-03-07 06:26:09 Is this correct? [Y/n]: /usr/local/bin/wspr: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.32' not found (required by /usr/local/bin/wspr)
2024-03-07 06:26:09 /usr/local/bin/wspr: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.34' not found (required by /usr/local/bin/wspr)
2024-03-07 06:26:09 /usr/local/bin/wspr: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /usr/local/bin/wspr)
2024-03-07 06:26:09 /usr/local/bin/wspr: /lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/local/bin/wspr)
2024-03-07 06:26:55 ( vs. 1.2.1-Alpha.1). Upgrade to newest? [Y/n]: (1.2.1-Alpha.1 vs. 1.2.1-Alpha.1). Overwrite anyway? [y/N]:
2024-03-07 06:26:55 Stopping wspr daemon.
2024-03-07 06:26:55 Disabling wspr daemon.
2024-03-07 06:26:55 Removed /etc/systemd/system/multi-user.target.wants/wspr.service.
2024-03-07 06:26:56 Removing unit file /etc/systemd/system/wspr.service
2024-03-07 06:26:56 Creating unit file for wspr (/etc/systemd/system/wspr.service).
2024-03-07 06:26:56 Reloading systemd config.
2024-03-07 06:26:57 Enabling wspr daemon.
2024-03-07 06:26:57 Created symlink /etc/systemd/system/multi-user.target.wants/wspr.service → /etc/systemd/system/wspr.service.
2024-03-07 06:26:58 Starting wspr daemon.
2024-03-07 06:27:01 Configuration file exists, overwrite? [y/N]:
2024-03-07 06:27:15 Support system shutdown button (TAPR)? [y/N]: (1.2.1-Alpha.1 vs. 1.2.1-Alpha.1). Overwrite anyway? [y/N]: (1.2.1-Alpha.1 vs. 1.2.1-Alpha.1). Overwrite anyway? [y/N]:
2024-03-07 06:27:15 Stopping shutdown-watch daemon.
2024-03-07 06:27:16 Disabling shutdown-watch daemon.
2024-03-07 06:27:16 Removed /etc/systemd/system/multi-user.target.wants/shutdown-watch.service.
2024-03-07 06:27:16 Removing unit file /etc/systemd/system/shutdown-watch.service
2024-03-07 06:27:17 Creating unit file for shutdown-watch (/etc/systemd/system/shutdown-watch.service).
2024-03-07 06:27:17 Reloading systemd config.
2024-03-07 06:27:17 Enabling shutdown-watch daemon.
2024-03-07 06:27:17 Created symlink /etc/systemd/system/multi-user.target.wants/shutdown-watch.service → /etc/systemd/system/shutdown-watch.service.
2024-03-07 06:27:18 Starting shutdown-watch daemon.
2024-03-07 06:27:24 (1.2.1-Alpha.1 vs. 1.2.1-Alpha.1). Overwrite anyway? [y/N]: Creating logrotate.d configuration.
2024-03-07 06:27:25 Updating any expired apt keys.
2024-03-07 06:27:25 Fixing any broken installations.
2024-03-07 06:27:25 Reading package lists...
2024-03-07 06:27:26 Building dependency tree...
2024-03-07 06:27:26 Reading state information...
2024-03-07 06:27:28 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2024-03-07 06:27:28 Checking and installing required dependencies via apt.
2024-03-07 06:27:31 Deleting any deprecated files.
2024-03-07 06:27:31 Checking and installing web pages.
2024-03-07 06:27:47 Fixing file permissions for /var/www/html/wspr.
2024-03-07 06:27:47 Fixing file permissions for data file.
2024-03-07 06:27:47 ***Script install.sh complete.***
2024-03-07 06:27:48 The WSPR daemon has started.
2024-03-07 06:27:48 - WSPR frontend URL : http://192.168.88.204/wspr
2024-03-07 06:27:48 -or- : http://WSPRPi.local/wspr
2024-03-07 06:27:48 - Release version : 1.2.1-Alpha.1
2024-03-07 06:27:48 Happy DXing!
```

```
pi@WSPRPi:~ $ sudo systemctl status wspr
● wspr.service - wspr daemon for: wspr
     Loaded: loaded (/etc/systemd/system/wspr.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2024-03-07 06:29:55 MST; 3s ago
       Docs: https://github.com/lbussy/WsprryPi/discussions
    Process: 683 ExecStart=/usr/local/bin/wspr -D -i /usr/local/etc/wspr.ini (code=exited, status=1/FAILURE)
   Main PID: 683 (code=exited, status=1/FAILURE)
        CPU: 4ms

Mar 07 06:30:00 WSPRPi systemd[1]: wspr.service: Scheduled restart job, restart counter is at 2.
Mar 07 06:30:00 WSPRPi systemd[1]: Stopped wspr daemon for: wspr.
Mar 07 06:30:00 WSPRPi systemd[1]: Started wspr daemon for: wspr.
Mar 07 06:30:04 WSPRPi systemd[1]: wspr.service: Main process exited, code=exited, status=1/FAILURE
Mar 07 06:30:04 WSPRPi systemd[1]: wspr.service: Failed with result 'exit-code'.
```

lbussy, 2024-03-07:
Maybe I should take up skydiving.

lbussy, 2024-03-07:
I love bungee jumping!  Skydiving too.  Maybe this explains ...

AA7US, 2024-03-08:
I finally have some good news Lee!

I downloaded a copy of 32 bit Raspberry Pi OS **Bookworm** this morning (recall that I've been using 32 bit Bullseye), installed it on an SD card, run OS updates, rebooted, then ran the WsprryPi alpha script.  I rebooted and found the wspr daemon was running!

Now everything is **almost** functioning perfectly with WsprryPi except: Beacon timing IS being honored when run from the command line but not when setup using the web GUI.  I think you already knew about this and are working on it.

FYI: Immediately after installing Bookworm, I ran `ldd ldd --version` and `strings $(/sbin/ldconfig -p | grep stdc++|awk -F'=>' '{print $2}')|grep LIBCXX` from the command line and now get the same results as you do... i.e.: the same versions are reported.

So I guess this means WsppryPi isn't supported running under Raspberry Pi OS Bullseye (Legacy) but is fine when running under Raspberry Pi OS Bookworm?

73,

John L.

lbussy, 2024-03-08:
Well, it all depends on whether I can figure out why it's not working.

Right now, I'm pissed at it, but not given up yet.  I'm VERY glad to hear you got it working, at least to where it's working here.

The app is small and I have always intended that it could be dropped onto any Pi, but maybe that's not practical with the low-level memory and GPIO control.

Also, 10M is dead today, so I have been fighting with this thing... :)

lbussy, 2024-03-09:
So it "looks like" we're going to have to live with "Bookworm or better" for the glibc errors.

I'm gonna just tell myself that's fine, maybe I come back to it, maybe now.

I will now go back to the original issue which was the web UI freq list not being honored.

AA7US, 2024-08-25:
Hi Lee,

I've been using WsprryPi exclusively from the command line to get beacon timing control since timing control wasn’t working for me from the web interface.  Today I decided to revisit the web interface just out of curiosity.  I setup the configuration to transmit a beacon every 8 minutes using “30m 0 0 0” and now it’s working perfectly!  IOW, the timing IS being honored from the web interface.  I don’t what’s changed since February when I couldn’t get it to work, but it’s fine now.

Come to think of it, there IS one thing I did… I recently reinstalled the software to enable the shutdown button option that I'd previously had disabled.  Maybe it's just coincidence or maybe that fixed something .  Anyway, I'm still using the alpha release version that you'd provided in one of these threads.

Thanks for software.  I use it all the time.

loschwa, 2024-10-15:
Dear Lee,
I recently came across your WSPR-software which is superb and exactly what I was looking for. Until now I was using for my WSPR-beacon the fork from James Peroulas which works fine but needs some tricks to be used standalone. Many thanks for making available you excellent software!
I did come across the same (or very similar) behavior which you and John had observed. I have installed WsprryPi on a RaspberryPi 2B running Kernel version 6.1.21-v7+. I used the web-interface for configuring WsprryPi. Everything worked as expected when I saved the configuration file the first time. I entered a single frequency using the Frequency Designator "6m". After checking that the transmission occurred at the correct frequency I entered the frequency string "6m 6m 6m 6m". I did that because I wanted to check the behavior with multiple frequency entries but didn't want to to tune my receiver for every frequency change. That worked as expected. Then I changed the frequency string to "2m 6m 6m". After saving the transmit log showed:

'2024-10-15 13:51:52 GMT	Notice: INI file changed, reloading parameters.
2024-10-15 13:51:52 GMT	Config reloaded from: /usr/local/etc/wspr.ini
2024-10-15 13:51:52 GMT	Transmit Enabled: true
2024-10-15 13:51:52 GMT	Call Sign: DK9FQ
2024-10-15 13:51:52 GMT	Grid Square: JN49
2024-10-15 13:51:52 GMT	Transmit Power: 10
2024-10-15 13:51:52 GMT	Frequencies: 2m 6m 6m
2024-10-15 13:51:52 GMT	PPM Offset: 0.000000
2024-10-15 13:51:52 GMT	Do not use NTP sync: false
2024-10-15 13:51:52 GMT	Check NTP Each Run (default): true
2024-10-15 13:51:52 GMT	Use Frequency Randomization: true
2024-10-15 13:51:52 GMT	Power Level: 3
2024-10-15 13:51:52 GMT	Use LED: false
2024-10-15 13:51:52 GMT	WSPR packet payload:
2024-10-15 13:51:52 GMT	- Callsign: DK9FQ
2024-10-15 13:51:52 GMT	- Locator: JN49
2024-10-15 13:51:52 GMT	- Power: 10 dBm
2024-10-15 13:51:52 GMT	Requested TX frequencies:
2024-10-15 13:51:52 GMT	- 144.490500 MHz - 50.294500 MHz - 50.294500 MHz
2024-10-15 13:51:52 GMT	Extra options:
2024-10-15 13:51:52 GMT	- Using NTP to calibrate transmission frequency. - A small random frequency offset will be added to all transmissions.
2024-10-15 13:51:52 GMT	Ready to transmit (setup complete).
2024-10-15 13:51:52 GMT	Center frequency for WSPR trans: 144.490469 MHz.
2024-10-15 13:51:52 GMT	Waiting for next WSPR transmission window.'

Which was as expected. When the WSPR transmission window opened the transmission started and the log was appended with: 

'2024-10-15 13:52:01 GMT	Obtained new ppm value: -3.658569
2024-10-15 13:52:01 GMT	Transmission started.
2024-10-15 13:53:51 GMT	Transmission completed, (110.611336 sec.)
2024-10-15 13:53:51 GMT	Center frequency for WSPR trans: 50.294509 MHz.
2024-10-15 13:53:51 GMT	Waiting for next WSPR transmission window.
2024-10-15 13:54:01 GMT	Transmission started.
2024-10-15 13:55:51 GMT	Transmission completed, (110.614037 sec.)
2024-10-15 13:55:51 GMT	Center frequency for WSPR trans: 50.294561 MHz.
2024-10-15 13:55:51 GMT	Waiting for next WSPR transmission window.
2024-10-15 13:56:01 GMT	Transmission started.
2024-10-15 13:57:51 GMT	Transmission completed, (110.612795 sec.)
2024-10-15 13:57:51 GMT	Center frequency for WSPR trans: 50.294496 MHz.
2024-10-15 13:57:51 GMT	Waiting for next WSPR transmission window.
2024-10-15 13:58:01 GMT	Transmission started.
2024-10-15 13:59:51 GMT	Transmission completed, (110.612401 sec.)
2024-10-15 13:59:51 GMT	Center frequency for WSPR trans: 144.490503 MHz.
2024-10-15 13:59:51 GMT	Waiting for next WSPR transmission window.
2024-10-15 14:00:01 GMT	Transmission started.'

As one can see, the first three transmissions happened as expected but then a further one occurred on 6m. After the completion of this transmission the cycle started again with a 2m transmission.

It appears to me that the original frequency string (6m 6m 6m 6m) was overwritten with the new one (2m 6m 6m) but did NOT use the proper string length of 8 characters but the old one of 11 characters. Hence another 6m transmission was performed before it recycled with a 2m transmission. 

Could it be that the pointer to the end of the frequency string is not properly updated when the new configuration was saved?

Please let me know if you need some clarification or further tests.

73s, Lothar (DK9FQ)

lbussy, 2024-10-15:
Grüß Lothar!

Thank you for those observations.  I have to admit, and I am ashamed to do so, that I have not even been able to install my dev environment on my new systems, let alone look at this.  The idea of grinding through this to find a place to start feeding my inner procrastination gene means I have not gotten anywhere on this.  You may have given me the foothold I need.

lbussy, 2024-11-19:
> As one can see, the first three transmissions happened as expected but then a further one occurred on 6m. After the completion of this transmission the cycle started again with a 2m transmission. [...]

That was it exactly.  Here is a log showing a fix to what you pointed out:

Starting with `20m 20m 0`:
```
2024-11-19 19:36:41 GMT Wsprry Pi v1.2.1-b2ae04b-dirty (fix_57).
2024-11-19 19:36:41 GMT Running on: Raspberry Pi 2B or 3B (BCM2837).
2024-11-19 19:36:41 GMT Config loaded from: /usr/local/etc/wspr.ini
2024-11-19 19:36:41 GMT Transmit Enabled: true
2024-11-19 19:36:41 GMT Call Sign: AA0NT
2024-11-19 19:36:41 GMT Grid Square: EM18
2024-11-19 19:36:41 GMT Transmit Power: 20
2024-11-19 19:36:41 GMT Frequencies: 20m 20m 0
2024-11-19 19:36:41 GMT PPM Offset: 0.000000
2024-11-19 19:36:41 GMT Do not use NTP sync: false
2024-11-19 19:36:41 GMT Check NTP Each Run (default): true
2024-11-19 19:36:41 GMT Use Frequency Randomization: true
2024-11-19 19:36:41 GMT Power Level: 7
2024-11-19 19:36:41 GMT Use LED: true
2024-11-19 19:36:41 GMT WSPR packet payload:
2024-11-19 19:36:41 GMT - Callsign: AA0NT
2024-11-19 19:36:41 GMT - Locator: EM18
2024-11-19 19:36:41 GMT - Power: 20 dBm
2024-11-19 19:36:41 GMT Requested TX frequencies:
2024-11-19 19:36:41 GMT - 14.097100 MHz - 14.097100 MHz - 0.000000 MHz
2024-11-19 19:36:41 GMT Extra options:
2024-11-19 19:36:41 GMT - Using NTP to calibrate transmission frequency. - A small random frequency offset will be added to all transmissions.
2024-11-19 19:36:41 GMT Ready to transmit (setup complete).
2024-11-19 19:36:41 GMT Center frequency for WSPR trans: 14.097156 MHz.
2024-11-19 19:36:41 GMT Waiting for next WSPR transmission window.
2024-11-19 19:38:01 GMT Obtained new ppm value: 12.026154
2024-11-19 19:38:01 GMT Transmission started.
2024-11-19 19:39:51 GMT Transmission completed, (110.612587 sec.)
2024-11-19 19:39:51 GMT Center frequency for WSPR trans: 14.097102 MHz.
2024-11-19 19:39:51 GMT Waiting for next WSPR transmission window.
2024-11-19 19:40:01 GMT Transmission started.
2024-11-19 19:41:51 GMT Transmission completed, (110.611699 sec.)
2024-11-19 19:41:51 GMT Center frequency for WSPR trans: 0.000000 MHz.
2024-11-19 19:41:51 GMT Waiting for next WSPR transmission window.
2024-11-19 19:42:01 GMT Obtained new ppm value: 9.305771
2024-11-19 19:42:01 GMT Skipping transmission.
2024-11-19 19:42:02 GMT Center frequency for WSPR trans: 14.097176 MHz.
2024-11-19 19:42:02 GMT Waiting for next WSPR transmission window.
2024-11-19 19:44:01 GMT Transmission started.
```

This works as expected.

Here, I updated to a new (shorter) frequency string of `0 20m` after saving one that was `20m 20m 20m`:
```
2024-11-19 19:54:25 GMT Notice: INI file changed, reloading parameters.
2024-11-19 19:54:25 GMT Config reloaded from: /usr/local/etc/wspr.ini
2024-11-19 19:54:25 GMT Transmit Enabled: true
2024-11-19 19:54:25 GMT Call Sign: AA0NT
2024-11-19 19:54:25 GMT Grid Square: EM18
2024-11-19 19:54:25 GMT Transmit Power: 20
2024-11-19 19:54:25 GMT Frequencies: 20m 20m 20m
2024-11-19 19:54:25 GMT PPM Offset: 0.000000
2024-11-19 19:54:25 GMT Do not use NTP sync: false
2024-11-19 19:54:25 GMT Check NTP Each Run (default): true
2024-11-19 19:54:25 GMT Use Frequency Randomization: true
2024-11-19 19:54:25 GMT Power Level: 7
2024-11-19 19:54:25 GMT Use LED: true
2024-11-19 19:54:25 GMT WSPR packet payload:
2024-11-19 19:54:25 GMT - Callsign: AA0NT
2024-11-19 19:54:25 GMT - Locator: EM18
2024-11-19 19:54:25 GMT - Power: 20 dBm
2024-11-19 19:54:25 GMT Requested TX frequencies:
2024-11-19 19:54:25 GMT - 14.097100 MHz - 14.097100 MHz - 14.097100 MHz
2024-11-19 19:54:25 GMT Extra options:
2024-11-19 19:54:25 GMT - Using NTP to calibrate transmission frequency. - A small random frequency offset will be added to all transmissions.
2024-11-19 19:54:25 GMT Ready to transmit (setup complete).
2024-11-19 19:54:25 GMT Center frequency for WSPR trans: 14.097085 MHz.
2024-11-19 19:54:25 GMT Waiting for next WSPR transmission window.
2024-11-19 19:54:33 GMT Notice: INI file changed, reloading parameters.
2024-11-19 19:54:33 GMT Config reloaded from: /usr/local/etc/wspr.ini
2024-11-19 19:54:33 GMT Transmit Enabled: true
2024-11-19 19:54:33 GMT Call Sign: AA0NT
2024-11-19 19:54:33 GMT Grid Square: EM18
2024-11-19 19:54:33 GMT Transmit Power: 20
2024-11-19 19:54:33 GMT Frequencies: 0 20m
2024-11-19 19:54:33 GMT PPM Offset: 0.000000
2024-11-19 19:54:33 GMT Do not use NTP sync: false
2024-11-19 19:54:33 GMT Check NTP Each Run (default): true
2024-11-19 19:54:33 GMT Use Frequency Randomization: true
2024-11-19 19:54:33 GMT Power Level: 7
2024-11-19 19:54:33 GMT Use LED: true
2024-11-19 19:54:33 GMT WSPR packet payload:
2024-11-19 19:54:34 GMT - Callsign: AA0NT
2024-11-19 19:54:34 GMT - Locator: EM18
2024-11-19 19:54:34 GMT - Power: 20 dBm
2024-11-19 19:54:34 GMT Requested TX frequencies:
2024-11-19 19:54:34 GMT - 0.000000 MHz - 14.097100 MHz
2024-11-19 19:54:34 GMT Extra options:
2024-11-19 19:54:34 GMT - Using NTP to calibrate transmission frequency. - A small random frequency offset will be added to all transmissions.
2024-11-19 19:54:34 GMT Ready to transmit (setup complete).
2024-11-19 19:54:34 GMT Center frequency for WSPR trans: 0.000000 MHz.
2024-11-19 19:54:34 GMT Waiting for next WSPR transmission window.
```

It picked up the new string when it reloaded the INI file.

I will do more testing, but I think this squashed the bug.

lbussy, 2025-02-07:
I'd tell you how crazy things have been, but that assumes my crazy is any different. :)

Did that change in November fix it?

AA7US, 2025-02-08:
Unfortunately beacon timing from the web console is still misbehaving for me.  Here's an example:

I first setup beaconing using the sequence "10M 0 0 0 0 0 0" on the web console and clicked save. That seemed to work fine with a beacon transmitted followed by the requested skips. After waiting awhile I changed the sequence to "10M 0" on the web console and clicked save again, but the new "10M 0" sequence isn't honored.

Here's the original wspr.ini file:

[Control]
Transmit = true

[Common]
Call Sign = AA7US
Grid Square = DM41
TX Power = 23
Frequency = 10m 0 0 0 0 0 0

[Extended]
PPM = 0
Power Level = 7
Self Cal = false
Offset = true
Use LED = true

Here's the changed wspr.ini file:

[Control]
Transmit = true

[Common]
Call Sign = AA7US
Grid Square = DM41
TX Power = 23
Frequency = 10m 0

[Extended]
PPM = 0
Power Level = 7
Self Cal = false
Offset = true
Use LED = true

Here's an excerpt from the transmit log. Note how the new "10M 0" sequence (highlighted in bold) is being ignored in the actual transmit sequence? The transmit sequence is still using the original "10M 0 0 0 0 0 0".

2025-02-08 18:47:52 GMT Notice: INI file changed, reloading parameters.
2025-02-08 18:47:52 GMT Config reloaded from: /usr/local/etc/wspr.ini
2025-02-08 18:47:52 GMT Transmit Enabled: true
2025-02-08 18:47:52 GMT Call Sign: AA7US
2025-02-08 18:47:52 GMT Grid Square: DM41
2025-02-08 18:47:52 GMT Transmit Power: 23
2025-02-08 18:47:52 GMT **Frequencies: 10m 0**
2025-02-08 18:47:52 GMT PPM Offset: 0.000000
2025-02-08 18:47:52 GMT Do not use NTP sync: true
2025-02-08 18:47:52 GMT Check NTP Each Run (default): false
2025-02-08 18:47:52 GMT Use Frequency Randomization: true
2025-02-08 18:47:52 GMT Power Level: 7
2025-02-08 18:47:52 GMT Use LED: true
2025-02-08 18:47:52 GMT WSPR packet payload:
2025-02-08 18:47:52 GMT - Callsign: AA7US
2025-02-08 18:47:52 GMT - Locator: DM41
2025-02-08 18:47:52 GMT - Power: 23 dBm
2025-02-08 18:47:52 GMT Requested TX frequencies:
2025-02-08 18:47:52 GMT - 28.126100 MHz - 0.000000 MHz
2025-02-08 18:47:52 GMT Extra options:
2025-02-08 18:47:52 GMT - A small random frequency offset will be added to all transmissions.
2025-02-08 18:47:52 GMT Ready to transmit (setup complete).
2025-02-08 18:47:52 GMT Center frequency for WSPR trans: 28.126135 MHz.
2025-02-08 18:47:52 GMT Waiting for next WSPR transmission window.
2025-02-08 18:48:01 GMT Transmission started.
2025-02-08 18:49:51 GMT Transmission completed, (110.625569 sec.)
2025-02-08 18:49:51 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 18:49:51 GMT Waiting for next WSPR transmission window.
2025-02-08 18:50:01 GMT Skipping transmission.
2025-02-08 18:50:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 18:50:02 GMT Waiting for next WSPR transmission window.
2025-02-08 18:52:01 GMT Skipping transmission.
2025-02-08 18:52:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 18:52:02 GMT Waiting for next WSPR transmission window.
2025-02-08 18:54:01 GMT Skipping transmission.
2025-02-08 18:54:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 18:54:02 GMT Waiting for next WSPR transmission window.
2025-02-08 18:56:01 GMT Skipping transmission.
2025-02-08 18:56:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 18:56:02 GMT Waiting for next WSPR transmission window.
2025-02-08 18:58:01 GMT Skipping transmission.
2025-02-08 18:58:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 18:58:02 GMT Waiting for next WSPR transmission window.
2025-02-08 19:00:01 GMT Skipping transmission.
2025-02-08 19:00:02 GMT Center frequency for WSPR trans: 28.126104 MHz.
2025-02-08 19:00:02 GMT Waiting for next WSPR transmission window.
2025-02-08 19:02:01 GMT Transmission started.
2025-02-08 19:03:51 GMT Transmission completed, (110.616253 sec.)
2025-02-08 19:03:51 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 19:03:51 GMT Waiting for next WSPR transmission window.
2025-02-08 19:04:01 GMT Skipping transmission.
2025-02-08 19:04:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 19:04:02 GMT Waiting for next WSPR transmission window.
2025-02-08 19:06:01 GMT Skipping transmission.
2025-02-08 19:06:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 19:06:02 GMT Waiting for next WSPR transmission window.
2025-02-08 19:08:01 GMT Skipping transmission.
2025-02-08 19:08:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 19:08:02 GMT Waiting for next WSPR transmission window.
2025-02-08 19:10:01 GMT Skipping transmission.
2025-02-08 19:10:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 19:10:02 GMT Waiting for next WSPR transmission window.
2025-02-08 19:12:01 GMT Skipping transmission.
2025-02-08 19:12:02 GMT Center frequency for WSPR trans: 0.000000 MHz.
2025-02-08 19:12:02 GMT Waiting for next WSPR transmission window.

lbussy, 2025-02-08:
I appreciate you looking. Mainly was just me hoping that I magically fixed it. I'll leave it on the list for what I'm working through now.

lbussy, 2025-02-27:
I am cut/pasting this in several issues that I've resolved in devel:

This iteration of Wsprry Pi became a total re-write.  The code I inherited was patched together over time and I finally reached the point where I needed to start over in order to make it work well.  Nothing in WSPR changed, of course, and the transmission is still the transmission, it's the management of the transmissions surrounding it that were a throw-away.

The good news is I am about 90% there, so keep your eyes peeled for a new release.

I need to ask someone I know who does IP law about the licensing for this project. The original was licensed under GNU GPL, and I have a fundamental distaste for GPL. I release all of my projects under the MIT license; I may have to release the new code in a new repo—if I do, I will announce it appropriately here.

lbussy, 2025-03-28:
Found it, squashed, officially in the new version for release when I am done. :)

lbussy, 2025-05-04:
Beta available.  First, cleanup:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/old_version_cleanup.sh  | sudo bash
```

then:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/install.sh  | sudo bash
```

AA7US, 2025-05-07:
This failure to honor timing is still present in 2.0 beta.

Here's a excerpt from wsprrypi.ini:

```
[Common]
; Call Sign: Your ham radio call sign (maximum 7 characters).
Call Sign = AA7US
; Grid Square: Your location's Maidenhead grid square (4 characters).
Grid Square = DM41
; TX Power: Transmitter power in dBm (integer, e.g., 20).
TX Power = 27
; Frequency: Transmission frequency in meters (e.g., '20m') or Hz (e.g., '450000000').
Frequency = 30m 0 0 0 0 0 0 0
; Transmit Pin: The GPIO pin for WSPR transmissions (BCM numbering).
Transmit Pin = 4
```

Here's a expert from wsprrypi.transmit.log:

```
pi@WSPRPi:/var/log/wsprrypi $ tail -n 20 wsprrypi.transmit.log
2025-05-07 01:05:29.791 UTC     [INFO ] - Callsign: AA7US
2025-05-07 01:05:29.791 UTC     [INFO ] - Locator: DM41
2025-05-07 01:05:29.791 UTC     [INFO ] - Power: 27 dBm
2025-05-07 01:05:29.791 UTC     [INFO ] Requested TX frequencies:
2025-05-07 01:05:29.791 UTC     [INFO ] - 10.140200 MHz
2025-05-07 01:05:29.791 UTC     [INFO ] - Skip (0.0)
2025-05-07 01:05:29.791 UTC     [INFO ] - Skip (0.0)
2025-05-07 01:05:29.791 UTC     [INFO ] - Skip (0.0)
2025-05-07 01:05:29.791 UTC     [INFO ] - Skip (0.0)
2025-05-07 01:05:29.791 UTC     [INFO ] - Skip (0.0)
2025-05-07 01:05:29.791 UTC     [INFO ] - Skip (0.0)
2025-05-07 01:05:29.791 UTC     [INFO ] - Skip (0.0)
2025-05-07 01:05:29.791 UTC     [INFO ] A random offset will be added to all transmissions.
2025-05-07 01:05:29.820 UTC     [INFO ] Setup complete, waiting for next transmission window.
2025-05-07 01:05:29.820 UTC     [INFO ] WSPR loop running.
2025-05-07 01:06:01.000 UTC     [INFO ] Transmission started.
2025-05-07 01:07:51.629 UTC     [INFO ] Transmission complete (110.628 secs).
2025-05-07 01:08:01.000 UTC     [INFO ] Transmission started.
2025-05-07 01:09:51.629 UTC     [INFO ] Transmission complete (110.629 secs).
2025-05-07 01:10:01.000 UTC     [INFO ] Transmission started.
```

It should transmit once then skip 7 cycles but instead keeps transmitting every cycle.

If it helps, in versions 1.2.2 and 1.2.3 it would correctly honor the timing after saving the configuration followed by a reboot, but not immediately (before a reboot). Now in 2.0 beta I can't get it to honor the timing with or without a reboot.

lbussy, 2025-05-07:
Can you verify the version reports as 2.0?  I tested this part pretty well.

AA7US, 2025-05-07:
> Can you verify the version reports as 2.0? I tested this part pretty well.

Yes...

```
pi@WSPRPi:~ $ sudo wsprrypi -v
WsprryPi version 2.0_Beta.2-heads-2-0-Beta-2 (heads-2-0-Beta-2).
```
I haven't yet tried running 2.0 beta from the CLI, but will give that a try tonight. On previous versions, running from the CLI the correct timing was always honored.

73,

John L.
AA7US

AA7US, 2025-05-07:
Another thing I want to try is a fresh install of Bookworm, then I'll install the wsprrypi 2.0 beta again to see if that makes any difference.

lbussy, 2025-05-07:
John, everyone loves and hates the guy who finds all the bugs.   :)

Let me give it a go.  This code was completely written from scratch, so I am amazed I managed to break it in the same way.

lbussy, 2025-05-07:
I found it.  I got too cute with my logic.  Should be fixed here shortly.

lbussy, 2025-05-07:
Incidentally, I realized why I never saw your issues/replies in the GitHub notifications.  I collaborate with another gentleman on some Homebrew (like, beer) projects, his name is also John, and I probably deleted the messages thinking it's a note in one of his projects that I was ignoring. :)

So I found the issue, here's my log emulating your setup (albeit at 40m):

```text
2025-05-07 19:10:53.200 UTC     [INFO ] WsprryPi version 2.0_Beta.2-fix-freq-iter+9efe41f (fix-freq-iter).
2025-05-07 19:10:53.200 UTC     [INFO ] Running on: Raspberry Pi 4 Model B Rev 1.1.
2025-05-07 19:10:53.200 UTC     [INFO ] Process PID: 13183
2025-05-07 19:10:53.201 UTC     [INFO ] Web server started on port: 31415
2025-05-07 19:10:53.201 UTC     [INFO ] WebSocketServer started on port: 31416
2025-05-07 19:10:53.201 UTC     [INFO ] WSPR loop running.
2025-05-07 19:10:53.201 UTC     [INFO ] PPM updated: 0.0
2025-05-07 19:10:53.342 UTC     [INFO ] WSPR packet payload:
2025-05-07 19:10:53.342 UTC     [INFO ] - Callsign: AA0NT
2025-05-07 19:10:53.342 UTC     [INFO ] - Locator: EM18
2025-05-07 19:10:53.342 UTC     [INFO ] - Power: 20 dBm
2025-05-07 19:10:53.342 UTC     [INFO ] Requested TX frequencies:
2025-05-07 19:10:53.342 UTC     [INFO ] - 7.040100 MHz
2025-05-07 19:10:53.342 UTC     [INFO ] - Skip (0.0)
2025-05-07 19:10:53.342 UTC     [INFO ] - Skip (0.0)
2025-05-07 19:10:53.342 UTC     [INFO ] - Skip (0.0)
2025-05-07 19:10:53.342 UTC     [INFO ] - Skip (0.0)
2025-05-07 19:10:53.342 UTC     [INFO ] - Skip (0.0)
2025-05-07 19:10:53.342 UTC     [INFO ] - Skip (0.0)
2025-05-07 19:10:53.342 UTC     [INFO ] - Skip (0.0)
2025-05-07 19:10:53.342 UTC     [INFO ] Using NTP to calibrate transmission frequency.
2025-05-07 19:10:53.401 UTC     [INFO ] A random offset will be added to all transmissions.
2025-05-07 19:10:53.428 UTC     [INFO ] DMA setup complete, waiting for next transmission window.
2025-05-07 19:12:01.000 UTC     [INFO ] Transmission started.
2025-05-07 19:13:51.588 UTC     [INFO ] Transmission complete (110.588 secs).
2025-05-07 19:13:51.623 UTC     [INFO ] DMA setup complete, waiting for next transmission window.
2025-05-07 19:14:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-07 19:16:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-07 19:18:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-07 19:20:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-07 19:22:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-07 19:24:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-07 19:26:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-07 19:26:01.037 UTC     [INFO ] DMA setup complete, waiting for next transmission window.
2025-05-07 19:28:01.000 UTC     [INFO ] Transmission started.
2025-05-07 19:29:51.588 UTC     [INFO ] Transmission complete (110.588 secs).
2025-05-07 19:29:51.622 UTC     [INFO ] DMA setup complete, waiting for next transmission window.
2025-05-07 19:29:55.835 UTC     [INFO ] Intercepted signal, shutdown will proceed: SIGTERM
2025-05-07 19:29:57.407 UTC     [INFO ] Web server stopped.
2025-05-07 19:29:57.408 UTC     [INFO ] WebSocketServer stopped.
2025-05-07 19:29:57.408 UTC     [INFO ] WsprryPi exiting.
```

I just need to figure out why, when it's not 0.0, the PPM is printing that, and I will publish another installer for you to try it out.

AA7US, 2025-05-07:
>...his name is also John, and I probably deleted the messages thinking it's a note in one of his projects that I was ignoring. :)

You should NEVER ignore a John! :)

> So I found the issue

Excellent!  I'm looking forward to testing it out.

73,

John L.
AA7US

lbussy, 2025-05-07:
Here ya go:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_Beta/scripts/install.sh | sudo bash
```

IN THEORY, the previous version cleanup is fixed too - so just let it rip.

AA7US, 2025-05-07:
Well Lee, I don't know what I've done to deserve this, but even with the new build you made, mine is still transmitting continuously. :(

I must've screwed something up somewhere, so tonight I'm going to try a fresh Bookworm build followed by a WsprryPi beta install without doing anything else.

Question:

Any idea why your transmit log shows this line:

[INFO ] DMA setup complete, waiting for next transmission window.

While mine doesn't?

Here's a sample of my transmit log:

```
pi@WSPRPi:/var/log/wsprrypi $ tail wsprrypi.transmit.log
2025-05-07 22:59:24.351 UTC     [INFO ] - Power: 27 dBm
2025-05-07 22:59:24.351 UTC     [INFO ] Requested TX frequencies:
2025-05-07 22:59:24.351 UTC     [INFO ] - 10.140200 MHz
2025-05-07 22:59:24.351 UTC     [INFO ] - Skip (0.0)
2025-05-07 22:59:24.351 UTC     [INFO ] - Skip (0.0)
2025-05-07 22:59:24.351 UTC     [INFO ] A random offset will be added to all transmissions.
2025-05-07 22:59:24.377 UTC     [INFO ] Setup complete, waiting for next transmission window.
2025-05-07 23:00:01.000 UTC     [INFO ] Transmission started.
2025-05-07 23:01:51.627 UTC     [INFO ] Transmission complete (110.627 secs).
2025-05-07 23:02:01.000 UTC     [INFO ] Transmission started.
```

Is my transmit log missing that line a clue to anything?

Also, after running the latest install script, should wsprrypi report the version as:

WsprryPi version 2.0_Beta.2-heads-2-0-Beta-2 (heads-2-0-Beta-2).

In my pi's home directory there is a file called "piped_script.log" and it shows this for the latest install:

```
2025-05-07 15:45:35 [INFO ] [piped_script:3202] Checking environment.
2025-05-07 15:45:36 [INFO ] [piped_script:3202] System: Raspbian GNU/Linux 12 (bookworm).
2025-05-07 15:45:36 [INFO ] [piped_script:3202] Running Wsprry Pi's install script, version 2.0_Beta.3.
2025-05-07 15:45:37 [INFO ] [piped_script:3202] Updating and managing required packages (this may take a few minutes).
2025-05-07 15:46:20 [INFO ] [piped_script:3202] APT package handling completed successfully.
2025-05-07 15:46:20 [INFO ] [piped_script:3202] Wsprry Pi installation beginning.
2025-05-07 15:46:20 [INFO ] [piped_script:3202] Ensuring destination directory does not exist: '/home/pi/WsprryPi'
2025-05-07 15:46:20 [INFO ] [piped_script:3202] Destination directory already exists: '/home/pi/WsprryPi'
2025-05-07 15:46:21 [INFO ] [piped_script:3202] Cleaning up older services.
2025-05-07 15:46:22 [INFO ] [piped_script:3202] Cleaning up older versions.
2025-05-07 15:46:23 [INFO ] [piped_script:3202] Installing wsprrypi.
2025-05-07 15:46:23 [INFO ] [piped_script:3202] Installing 'wsprrypi' configuration.
2025-05-07 15:46:27 [INFO ] [piped_script:3202] Updating systemd service: wsprrypi.service.
2025-05-07 15:46:34 [INFO ] [piped_script:3202] Systemd service wsprrypi created.
2025-05-07 15:46:34 [INFO ] [piped_script:3202] Installing 'logrotate' configuration.
2025-05-07 15:46:34 [INFO ] [piped_script:3202] Installing web files to '/var/www/html/wsprrypi'.
2025-05-07 15:46:35 [INFO ] [piped_script:3202] Deleting local repository tree.
```

Thanks,

Frustrated

lbussy, 2025-05-07:
I just fixed the "piped script" thing - try one more time before you blow things away and tell me what you get.  There should be a `wsprrypi_install.log` in your home directory after.

But I see that the executable did not copy over because it should be `2.0_Beta.3` so let's see what the log says.

lbussy, 2025-05-07:
And if that did not work, try this:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/do_not_use/scripts/install.sh | sudo bash
```

... pay no attention to the "do_not_use" in the URL :)

AA7US, 2025-05-08:
You nailed it Lee!

It's now working correctly after rerunning the (regular) beta installer script.

```
2025-05-08 00:15:44.257 UTC     [INFO ] WSPR packet payload:
2025-05-08 00:15:44.258 UTC     [INFO ] - Callsign: AA7US
2025-05-08 00:15:44.258 UTC     [INFO ] - Locator: DM41
2025-05-08 00:15:44.258 UTC     [INFO ] - Power: 27 dBm
2025-05-08 00:15:44.258 UTC     [INFO ] Requested TX frequencies:
2025-05-08 00:15:44.258 UTC     [INFO ] - 10.140200 MHz
2025-05-08 00:15:44.258 UTC     [INFO ] - Skip (0.0)
2025-05-08 00:15:44.258 UTC     [INFO ] - Skip (0.0)
2025-05-08 00:15:44.258 UTC     [INFO ] A random offset will be added to all transmissions.
2025-05-08 00:15:44.291 UTC     [INFO ] DMA setup complete, waiting for next transmission window.
2025-05-08 00:16:01.000 UTC     [INFO ] Transmission started (frequency = 10.140200 MHz).
2025-05-08 00:17:51.632 UTC     [INFO ] Transmission complete (110.631 secs).
2025-05-08 00:17:51.667 UTC     [INFO ] DMA setup complete, waiting for next transmission window.
2025-05-08 00:18:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-08 00:20:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-08 00:20:01.033 UTC     [INFO ] DMA setup complete, waiting for next transmission window.
2025-05-08 00:22:01.000 UTC     [INFO ] Transmission started (frequency = 10.140200 MHz).
```

The correct version number is also being reported:

WsprryPi version 2.0_Beta.3-2-0-Beta (2-0-Beta).

And the home directory now has a "wsprrypi.log" file.

Thank you, thank you, thank you!

John L.
AA7US

PS: With my luck I'm sure I'll create or find some new bugs. :)

[1 more reply]

closed 2025-05-08 by lbussy