WsprryPi/WsprryPi #58: Filesystem Error Using Command Line

AA7US, 2024-02-21:
Attempting to run some wspr beacons by using the command line and have encountered a new error:

**terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'
  what():  filesystem error: cannot get file time: No such file or directory []
Exiting, caught signal: 6**

The wspr daemon was stopped first.

Here's the command I used and the full result:

sudo /usr/local/bin/wspr -s -o -r AA7US DM41 13 10140200 0 0 0
Wsprry Pi v1.1.0 (main).
Running on: Raspberry Pi 2B or 3B (BCM2837).
WSPR packet payload:
- Callsign: AA7US
- Locator:  DM41
- Power:    13 dBm
Requested TX frequencies:
- 10.140200 MHz
- 0.000000 MHz
- 0.000000 MHz
- 0.000000 MHz

Extra options:
- Using NTP to calibrate transmission frequency.
- Transmissions will continue forever until stopped with CTRL-C.
- A small random frequency offset will be added to all transmissions.

Ready to transmit (setup complete).
Center frequency for WSPR trans: 10.140188 MHz.
Waiting for next WSPR transmission window.
terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'
  what():  filesystem error: cannot get file time: No such file or directory []
Exiting, caught signal: 6

lbussy, 2024-02-22:
Received - lemme poke at it

AA7US, 2024-02-23:
Could this error be caused by using a relative file path somewhere in the code instead of an absolute (full) file path?

lbussy, 2024-02-23:
Maybe.  I just need to get in there and look.  I have another minor issue with the supporting header files so it's going to take a quiet house and lots of coffee. 😄

lbussy, 2024-02-25:
Working on this today ... no promises but I have some quiet time.

lbussy, 2024-02-25:
This is fixed but not yet released.  The issue was that I was not checking if we were using an .ini file in the loop.

This fix will be in the next release (in theory 1.2.1.)

closed 2024-02-25 by lbussy