WsprryPi/WsprryPi #60: Check for Time Service

lbussy, 2024-02-25:
@edegraaff reports that when installed on an OD Pi Zero (no "W"), there is no time synch service installed, even if there's a dongle added.

Consider checking for a list of `ntpd`-ish services.

edegraaff, 2024-02-26:
> @edegraaff reports that when installed on an OG Pi Zero (no "W"), no time synch service is installed, even if a dongle is added.
> 
> Consider checking for a list of `ntpd`-ish services.

Hi Lee, i use the Pi Zero W version 1.1 and raspbian does install a time service. What service and how accurate i have no clue. That was the reason to make use of NTP where you can configure the location and the best time resource. for the netherlands is https://www.ntppool.org/en/zone/nl the proposed servers. 

Since time is quite important and most ham's do have some sort of timesource or in their house or are familiar with it, some suggestions:
In the webgui : the reboot button, and also the current time hh:mm:ss and if it is local or utc... it gives some guidance if the wspr is within the time bounderys. 
Some routers have internally a ntp server and there you could tapp into this service. 
I also realize implementation of ntp could make the install scripts more complex, and the project is now straight forward. 

Another suggestion is to show if the wspr service is transmitting, and when not, i have found some situations the beacon was invissible and after a reboot all worked as expexted. 

Just some suggestions.  

regards 
73 de pd1eg

lbussy, 2024-03-02:
On your Pi Zero, would you please check `timedatectl`?

edegraaff, 2024-03-02:
> On your Pi Zero, would you please check `timedatectl`?

Hi Lee i have made a clean instal. The timedatectl does work on the pi zero also when you install the base image. Het issue that was creating problems is when you install ntp. 
timedatectl status was broken in de previous install, new install ok ! 
timedatectl status
               Local time: Sat 2024-03-02 18:15:13 CET
           Universal time: Sat 2024-03-02 17:15:13 UTC
                 RTC time: n/a
                Time zone: Europe/Amsterdam (CET, +0100)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
pi@pizerowspr:~ $
Adding somewhere the current time and/or date could trigger users to check the clock. 

I added an extra ntp server in the config.
sudo nano /etc/systemd/timesyncd.conf
I am synced to a part of a second now. 
Regards Eelco de pd1eg

lbussy, 2024-03-03:
Well, that's good news - okay, it seems I don't need to find a Pi Zero to address this. :)

lbussy, 2025-03-28:
I am re-opening this as a tracker so there will be a notification when I do release.

I am using chrony in the new version, which seems to be reliable and also allows PPM corrections even when NTP is not available.

lbussy, 2025-05-04:
Beta available.  First, cleanup:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/old_version_cleanup.sh  | sudo bash
```

then:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/install.sh  | sudo bash
```

AA7US, 2025-05-07:
Lee,

When installing or updating wsprrypi, would it be possible to check if chrony already exists, and if it does, maybe not install it again... or at least not change /etc/chrony/chrony.conf?

I ask because I already had chrony installed and configured to use my own in-house GPS-disciplined NTP server, and when I installed the 2.0 beta, the existing /etc/chrony/chrony.conf was overwritten so my NTP server was no longer being used.

Thanks!

lbussy, 2025-05-07:
Whew!  :)

I only upgrade Chrony if it's there, just an apt function, so I would have had to blame it on the upstream.  Cool you found Chrony too, though.  I find it much easier and more reliable.  NTPD seems to be suffering from bloat.

closed 2025-05-07 by lbussy