WsprryPi/WsprryPi #62: Allow Reboot

lbussy, 2024-03-05:
@edegraaff suggested a reboot button.

lbussy, 2025-02-14:
This is a reminder for myself: There is a power button on the web page; add a second icon. Font Awesome does not (yet) have a likely icon; the issue is [here](https://github.com/FortAwesome/Font-Awesome/issues/19762).

lbussy, 2025-02-17:
Here's a good one:  https://fontawesome.com/icons/arrows-rotate

AA7US, 2025-02-17:
That’s perfect!

lbussy, 2025-02-17:
Getting there.  This branch is still buggy but will work (enough for Beta) when I merge to 1.3.0_devel.  Should be a little later today.

edegraaff, 2025-02-17:
we have no deadline grin

lbussy, 2025-02-17:
That's good, I ain't done!   :)

I got some security updates done (bootstrap and jquery) and got rid of some nagging issues in the previous UI.  It's a re-write but "good" looked like the old one and I did that. 

Letting the new executable burn in, have a look for AA0NT on 40m for signs of life.

lbussy, 2025-02-18:
Closing this one as complete, too, as it is in my dev version.  Here's what it looks like:

[image]

lbussy, 2025-02-27:
I am cut/pasting this in several issues that I've resolved in devel:

This iteration of Wsprry Pi became a total re-write.  The code I inherited was patched together over time and I finally reached the point where I needed to start over in order to make it work well.  Nothing in WSPR changed, of course, and the transmission is still the transmission, it's the management of the transmissions surrounding it that were a throw-away.

The good news is I am about 90% there, so keep your eyes peeled for a new release.

I need to ask someone I know who does IP law about the licensing for this project. The original was licensed under GNU GPL, and I have a fundamental distaste for GPL. I release all of my projects under the MIT license; I may have to release the new code in a new repo—if I do, I will announce it appropriately here.

lbussy, 2025-03-28:
I am re-opening this as a tracker so there will be a notification when I do release.

lbussy, 2025-05-04:
Beta available.  First, cleanup:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/old_version_cleanup.sh  | sudo bash
```

then:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/install.sh  | sudo bash
```

closed 2025-05-07 by lbussy