WsprryPi/WsprryPi #64: Show Transmission Indicator

lbussy, 2024-03-05:
@edegraaff suggested a transmission indicator.

lbussy, 2025-02-14:
A possible Font Awesome candidate: https://fontawesome.com/icons/radio?s=solid

This functionality requires better inter-process communication running in my 1.3.0-devel branch. I hope to support this with that release.

AA7US, 2025-02-14:
What about this icon?

https://fontawesome.com/icons/tower-cell?s=solid

lbussy, 2025-02-14:
Maybe.  That's already close to the favicon and logo I use. I was also thinking I could change the logo's color to red to denote transmission.

Good ideas often change as I try to implement them, so who knows? :)

lbussy, 2025-02-27:
This one may be less straightforward than it seems.  In WSPR mode, there are only 10 seconds in between transmissions.  That means there would need to be near real-time communication between the web page and the application.  While that's not unique in the world, it is a major change to how Wsprry Pi works right now.

I got the rest of the stuff we talked about, but this one may have to wait for a follow-up release.

lbussy, 2025-03-27:
Okay, I was able to incorporate this when I added a REST API as well as WebSockets.  Development, 10 minutes at a time, is tough, but I am down to one major class to re-write, and then it will be punch-list and testing.

Keep the faith!

AA7US, 2025-03-27:
Thank you for everything you’re doing Lee. You are appreciated.

AA7US

lbussy, 2025-05-04:
Beta available.  First, cleanup:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/old_version_cleanup.sh  | sudo bash
```

then:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/install.sh  | sudo bash
```

closed 2025-05-07 by lbussy