WsprryPi/WsprryPi #68: Script breaks

K3KDW, 2024-08-01:
When the script gets to the "Support system shutdown button..." line the ssh terminal into the pi freezes regardless of whether I select y or N. I'm running a pi zero (not W, want to use a nice external usb wifi dongle) with a fresh install of Pi OS legacy 32-bit.

I am working around this by removing lines 864-876:

    # Choose to support shutdown button
    no_tapr=""
    read -rp "Support system shutdown button (TAPR)? [y/N]: " yn  < /dev/tty
    case "$yn" in
        [Yy]* ) no_tapr="false";;
        [Nn]* ) no_tapr="true";;
        * ) no_tapr="true";;
    esac
    do_unit "shutdown-watch" "python3"
    # Optional: Turn off TAPR button handling
    if [ "$no_tapr" == "true" ]; then
        sed -i 's/^doTAPR = True/doTAPR = False/' /usr/local/bin/shutdown-watch.py
    fi

That completed the install and I got the "Heppy DXing!" message. Rebooted. Checked the webpage and it was just the standard "It works" Apache2 Debian default page.

I tried the script again and it now hangs (regardless of selection) on "Log rotate exists and is the same..." so I removed lines 441-463:

        if [ "$verchk" == "lt" ]; then
            echo -e "Log rotate exists but is an older version" > /dev/tty
            read -rp "($src vs. $VERSION). Upgrade to newest? [Y/n]: " yn < /dev/tty
            case "$yn" in
                [Nn]* )
                    retval="false";;
                * )
                    retval="true" ;; # Do overwrite
            esac
        elif [ "$verchk" == "eq" ]; then
            echo -e "\nLog rotate exists and is the same version" > /dev/tty
            read -rp "($src vs. $VERSION). Overwrite anyway? [y/N]: " yn < /dev/tty
            case "$yn" in
                [Yy]* )
                    retval="true";; # Do overwrite
                * )
                    retval="false";;
            esac
        elif [ "$verchk" == "gt" ]; then
            echo -e "\nLog rotate is newer than the version being installed."
            echo -e "Skipping."
            retval="false"
        fi

I updated, upgraded, and rebooted for good measure before running the further pared down script. It didn't work. I said no to upgrading wspr and overwriting files, etc. which again said it was all successful but upon restart the same Apache2 default page was in place. I can run wspr through an ssh connection but it freezes again.

Is it possible this is all from underpowering? I've tried a few power configurations through both the pwr and the usb otg port with no luck. Using otg for an ethernet connection and at various stages also a usb wifi dongle.

K3KDW, 2024-08-02:
After doing more research it looks like running a headless zero causes issues on various forks. Closing the issue and making a new plan.

lbussy, 2025-03-28:
Heh, this got opened and closed before I saw it.  The new version internalizes the LED and shutdown work (and allows configuration via INI ifile.)  I am going to re-open this because this is fixed in the new version by virtue of a new feature.

PR #132 (2025-04-06): Bugfixes 1.2.2
Closes #101 
Closes #68
closed 2025-04-06 by lbussy