WsprryPi/WsprryPi #69: Generate band switching control signals for frequency-hopping

loschwa, 2024-10-15:
Dear Lee,
it would be extremely useful if the RaspberryPi would in addition to the frequency generation on GPIO4 also provide a control signal on some of the other GPIOs for the selection  of an appropriate low-pass filter. With that feature it would be possible to use WsprryPi on multiple frequencies without manually switching the low-pass filters. For instance it would be possible to control the 'Ultimate relay-switched LPF kit' of QRP-Labs. This kit allows to select one of 5 different low-pass filters and it requires a logical '1' on either one of its 5 inputs.

This could work such that the first 5 frequency 'slots' in the frequency configuration string ( e.g. '80m 40m 30m 20m 15m') would be associated with GPIOs 17, 27, 22, 23, 24. When WsprryPi starts transmitting on 80m the GPIO17 outputs a '1' which would allow the selection of the 80m-low-pass filter. When the transmission stops the GPIO17-output reverts to '0'. When the transmission window opens for 40m, GPIO27 outputs a '1' ... and so on.

Please let me know if I could help to implement the new function. However, I don't have a profound knowledge of C-programming but I could help with the documentation.

73s, Lothar (DK9FQ)

lbussy, 2024-11-02:
Hi, Lothar.  I did not see this when you posted it originally.  I agree; that would be nice.  It was one of the things I viewed as a potential enhancement for the future.  As people do, this fell off my radar, and now that there are a couple of bugs I need to fix and a possible rewrite to take advantage of Pi 5 and 64-bit OS, I'm back in here sniffing around.

I'll mark this as an enhancement so I do not lose track.

lbussy, 2026-03-24:
If anyone wants to try this in an Alpha mode, let me know.  Right now it only works on these bands, active low, and with this pin alignment:

```cpp
    switch (band)
    {
        case HamBand::BAND_2200M: return 17;
        case HamBand::BAND_630M:  return 27;
        case HamBand::BAND_160M:  return 22;
        case HamBand::BAND_80M:   return 23;
        case HamBand::BAND_60M:   return 24;
        case HamBand::BAND_40M:   return 25;
        case HamBand::BAND_30M:   return 5;
        case HamBand::BAND_22M:   return 6;
        case HamBand::BAND_20M:   return 12;
        case HamBand::BAND_17M:   return 13;
        case HamBand::BAND_15M:   return 16;
        case HamBand::BAND_12M:   return 26;
        case HamBand::BAND_10M:   return 20;
        case HamBand::BAND_6M:    return 21;
    }
```

lbussy, 2026-03-28:
The configuration is static right now, but I will allow combinations when I put in a configuration page.

I am trying to get a new version of a board to the fab house that will work with the Pi 5, and then back to this.

5ft24dave, 2026-04-05:
Just did the git clone line manually, it downloads,  and at the end I get this error  WsprryPi-UI is empty with just the .git file in it

```
`Cloning into '/home/pi/WsprryPi/src/WSPR-Transmitter'...
remote: Enumerating objects: 744, done.
remote: Counting objects: 100% (229/229), done.
remote: Compressing objects: 100% (91/91), done.
remote: Total 744 (delta 169), reused 150 (delta 138), pack-reused 515 (from 1)
Receiving objects: 100% (744/744), 456.58 KiB | 4.08 MiB/s, done.
Resolving deltas: 100% (415/415), done.
fatal: remote error: upload-pack: not our ref 966e60bcfe76c7023580717b297539b0f6e571b0
fatal: Fetched in submodule path 'WsprryPi-UI', but it did not contain 966e60bcfe76c7023580717b297539b0f6e571b0. Direct fetching of that commit failed.
`
```

lbussy, 2026-04-05:
Best kind of news to wake up to!

lbussy, 2026-04-13:
That's a good report. I mean I hate that it's not working right but I'm happy to have such a clear problem statement, thank you.

lbussy, 2026-04-18:
Okay thanks for checking.  I'm home now so I can put my handy dandy LED board on there and see what happens.

lbussy, 2026-04-19:
Success!

lbussy, 2026-04-19:
New Beta, new UI.  See Issue #286 for details, and provide feedback on the UI in that issue only.

```bash
curl -fsSL https://raw.githubusercontent.com/WsprryPi/WsprryPi/refs/heads/devel/scripts/install.sh | sudo bash
```

WsprryPi version 3.0.0-beta.3+ui-updates-devel (devel).

[image]

5ft24dave, 2026-04-25:
I'm having issues... installed, then because of the issue reinstalled PiOS fresh and installed and same issue.  Go through all the config, go to operations page and get the attached... go back to setup and it's all blank again.  checked the /usr/lib/wsprry.ini file and it has all the info I entered.  rebooted multiple times and it still isn't keeping the config after I go to the operations page and back.

[image]

_**2026-04-25T19:11:17.054Z wsprrypi.service [INFO ] Configuration file repaired from stock.
2026-04-25T18:56:06.532Z init.scope       [INFO ] Started wsprrypi.service - WsprryPi daemon for: wsprrypi.
2026-04-25T18:56:06.658Z wsprrypi.service [INFO ] WsprryPi version 3.0.0-rc.1-devel+71fd1b3 (devel).
2026-04-25T18:56:06.663Z wsprrypi.service [INFO ] Platform: Raspberry Pi 4 Model B Rev 1.4.
2026-04-25T18:56:06.663Z wsprrypi.service [INFO ] OS: Debian GNU/Linux 13 (trixie) 64-bit.
2026-04-25T18:56:06.664Z wsprrypi.service [INFO ] Transmit backend: gpio
2026-04-25T18:56:06.664Z wsprrypi.service [INFO ] Transmit GPIO: 4
2026-04-25T18:56:06.664Z wsprrypi.service [INFO ] Web server started on port: 31415
2026-04-25T18:56:06.664Z wsprrypi.service [INFO ] Socket server started on port: 31416
2026-04-25T18:56:06.675Z wsprrypi.service [INFO ] WSPR loop running.
2026-04-25T18:56:06.675Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T18:56:06.805Z wsprrypi.service [INFO ] PPM Manager initialized successfully.
2026-04-25T18:56:06.805Z wsprrypi.service [INFO ] PPM updated: 0.071
2026-04-25T18:56:06.806Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T18:56:06.806Z wsprrypi.service [INFO ] Transmit disabled, skipping transmission and scheduling.
2026-04-25T18:57:57.324Z init.scope       [INFO ] Stopping wsprrypi.service - WsprryPi daemon for: wsprrypi...
2026-04-25T18:57:57.325Z wsprrypi.service [INFO ] Shutdown requested: received SIGTERM signal
2026-04-25T18:57:57.784Z wsprrypi.service [INFO ] WsprryPi exiting.
2026-04-25T18:57:57.799Z init.scope       [NOTIC] wsprrypi.service: Main process exited, code=exited, status=143/n/a
2026-04-25T18:57:57.800Z init.scope       [WARN ] wsprrypi.service: Failed with result 'exit-code'.
2026-04-25T18:57:57.801Z init.scope       [INFO ] Stopped wsprrypi.service - WsprryPi daemon for: wsprrypi.
2026-04-25T19:04:49.272Z init.scope       [INFO ] Started wsprrypi.service - WsprryPi daemon for: wsprrypi.
2026-04-25T19:04:49.320Z wsprrypi.service [INFO ] WsprryPi version 3.0.0-rc.1-devel+71fd1b3 (devel).
2026-04-25T19:04:49.321Z wsprrypi.service [INFO ] Platform: Raspberry Pi 4 Model B Rev 1.4.
2026-04-25T19:04:49.321Z wsprrypi.service [INFO ] OS: Debian GNU/Linux 13 (trixie) 64-bit.
2026-04-25T19:04:49.321Z wsprrypi.service [INFO ] Transmit backend: gpio
2026-04-25T19:04:49.321Z wsprrypi.service [INFO ] Transmit GPIO: 4
2026-04-25T19:04:49.321Z wsprrypi.service [INFO ] Web server started on port: 31415
2026-04-25T19:04:49.321Z wsprrypi.service [INFO ] Socket server started on port: 31416
2026-04-25T19:04:49.322Z wsprrypi.service [INFO ] WSPR loop running.
2026-04-25T19:04:49.322Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T19:04:49.378Z wsprrypi.service [INFO ] PPM Manager initialized successfully.
2026-04-25T19:04:49.378Z wsprrypi.service [INFO ] PPM updated: 0.071
2026-04-25T19:04:49.378Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T19:04:49.378Z wsprrypi.service [INFO ] Transmit disabled, skipping transmission and scheduling.
2026-04-25T19:05:36.451Z wsprrypi.service [INFO ] INI file changed, reloading.
2026-04-25T19:05:36.454Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T19:05:36.454Z wsprrypi.service [INFO ] Transmit backend: gpio
2026-04-25T19:05:36.454Z wsprrypi.service [INFO ] Transmit GPIO: 4
2026-04-25T19:05:37.372Z wsprrypi.service [INFO ] PPM Manager disabled.
2026-04-25T19:05:37.372Z wsprrypi.service [INFO ] Transmit disabled, skipping transmission and scheduling.
2026-04-25T19:05:38.928Z wsprrypi.service [INFO ] INI file changed, reloading.
2026-04-25T19:05:38.931Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T19:05:38.931Z wsprrypi.service [INFO ] Transmit backend: gpio
2026-04-25T19:05:38.932Z wsprrypi.service [INFO ] Transmit GPIO: 4
2026-04-25T19:05:38.932Z wsprrypi.service [INFO ] PPM Manager disabled.
2026-04-25T19:05:38.933Z wsprrypi.service [INFO ] Transmit disabled, skipping transmission and scheduling.
2026-04-25T19:05:42.227Z wsprrypi.service [INFO ] INI file changed, reloading.
2026-04-25T19:05:42.229Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T19:05:42.229Z wsprrypi.service [INFO ] Transmit backend: gpio
2026-04-25T19:05:42.229Z wsprrypi.service [INFO ] Transmit GPIO: 4
2026-04-25T19:05:42.229Z wsprrypi.service [INFO ] PPM Manager disabled.
2026-04-25T19:05:42.230Z wsprrypi.service [INFO ] Transmit disabled, skipping transmission and scheduling.
2026-04-25T19:05:46.226Z wsprrypi.service [INFO ] INI file changed, reloading.
2026-04-25T19:05:46.228Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T19:05:46.228Z wsprrypi.service [INFO ] Transmit backend: gpio
2026-04-25T19:05:46.229Z wsprrypi.service [INFO ] Transmit GPIO: 4
2026-04-25T19:05:46.229Z wsprrypi.service [INFO ] PPM Manager disabled.
2026-04-25T19:05:46.229Z wsprrypi.service [INFO ] Transmit disabled, skipping transmission and scheduling.
[... cut ...]
2026-04-25T19:10:56.171Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T19:10:56.232Z wsprrypi.service [INFO ] PPM Manager initialized successfully.
2026-04-25T19:10:56.232Z wsprrypi.service [INFO ] PPM updated: 0.062
2026-04-25T19:10:56.232Z wsprrypi.service [INFO ] Scheduling path selected: WSPR.
2026-04-25T19:10:56.232Z wsprrypi.service [INFO ] Transmit disabled, skipping transmission and scheduling.
2026-04-25T19:11:17.054Z wsprrypi.service [INFO ] Configuration file repaired from stock.**_

closed 2026-04-26 by lbussy