WsprryPi/WsprryPi #86: Feature Request: Add QRSS Functionality

AA7US, 2024-11-23:
Hi Lee,

[Fabian Kurz, DJ1YFK](https://www.qrz.com/db/DJ5CW?aliasFrom=DJ1YFK) has figured out how to add simple [QRSS](https://swharden.com/blog/2020-10-03-new-age-of-qrss/) beacon capability to one of the original WsspryPi forks.  This is done with a new "-q" option.  His repository is [HERE](https://git.fkurz.net/dj1yfk/WsprryPi).  Unfortunately, his version lacks the newer shutdown button and transmit LED features.

I would love to see QRSS functionality added to your version of WsprryPi.  That would make it the ultimate WSPR / QRSS dual mode application for the Pi.  Possibly the two modes could be easily integrated into the web console using a toggle to chose between either WSPR or QRSS.

I run Fabian's command line only version using a bash script that first executes a QRSS beacon on the 1's (1 minute after the hour, 11 minutes, 21 minute etc.) followed by a WSPR beacon on the 8's (8 minutes after the hour, 18 minutes, 28 minutes, etc.).  This kind of use may be too complicated for the web console.

lbussy, 2025-02-07:
Mebbe.  I'll have a look when I get a couple more things done and see what I can fit in.

lbussy, 2025-06-04:
Putting a [K4VRC/Scott Harden link](http://www.k4vrc.com/uploads/7/8/8/6/78865320/scott-harden-qrss-resources.pdf_ here to jog my memory.

lbussy, 2025-06-04:
Also a link to where I am piddling around (this is a private repo, sorry): [WSPR-QRSS](https://github.com/lbussy/WSPR-QRSS).

AA7US, 2025-07-09:
Nice! 😊

lbussy, 2025-07-10:
More teases:

[image]

[image]

I'm still tweaking the layout (and I will disable some fields depending on mode), but it's mostly how I want it.

AA7US, 2025-07-10:
Lookin' good! 👍🏼