WsprryPi/WsprryPi #92: Suggestion since there is a new version Thanks!

edegraaff, 2025-01-16:
If posible in a future release show the version of the software somewhere in the screen.
This makes it for me easy to see if i need to upgrade.
I did upgrade in place with the script, worked without any issues thanks!

regards
Eelco

lbussy, 2025-02-07:
Indeed - I was noodling on that today (and I need to do something about my notifications.). I put this in #97

Try out the VERY Beta version:

First, make sure your system is up to date.  If you are not on Bookworm, start with a new 32-bit image.  Then:

```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
```

Next, clean the older binaries (this is not yet in the installer):

`curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/1.3.0_devel/scripts/old_version_cleanup.sh | sudo bash`

Followed by:

`curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/1.3.0_devel/scripts/install.sh | sudo bash`

I would not recommend that anyone else try this unless they are stuck.

lbussy, 2025-02-08:
I'm leaving myself a note here to check into musl for linking libs.so.6 statically.

lbussy, 2025-02-13:
@edegraaff did that help?

lbussy, 2025-02-14:
In case anyone is reading this, GLIBC is a foundational library provided with the operating system. The issue we have here is that when I compile on, say, Bookwork and someone uses Bullseye, the libraries on the older system do not support the executable.

Static linking of GLIBC is discouraged and discouraging to try.  There are many pages [and threads](https://news.ycombinator.com/item?id=40358930) about GLIBC woes; I have a hard time believing this cannot be solved more readily.  

I may have to adopt either a local compile during installation or test out supporting the [oldest supported release](https://www.debian.org/releases/) and risk missing important GLIBC updates.  

This is a release problem, I'll worry about it when I get to that point.

edegraaff, 2025-02-14:
Since the configuration is quite easy, i would say don't put a lot of effort in it. Flashing a pi is done in 15 min with config. An alternative could be, export config to a json file and make a backup export import as alternative... with new builds, jou just point out latest raspbian build :-)

lbussy, 2025-02-14:
Generally speaking, folks who post an issue are more technically adept than those that quietly give up and walk away. 

I lead service delivery in my day job, so thinking through how I could address it as a customer no-touch event is interesting to me.

edegraaff, 2025-02-14:
grin,  seems we share some interests...

lbussy, 2025-02-18:
I'm closing this one with the understanding that one will have to use the current version of Raspbian to have the proper GLIBC headers. That's a "for now" because I can look into a cross-compiler later and do some CI/CD work. That's outside scope at the moment, so this one is officially not a bug.

lbussy, 2025-02-27:
I am cut/pasting this in several issues that I've resolved in devel:

This iteration of Wsprry Pi became a total re-write.  The code I inherited was patched together over time and I finally reached the point where I needed to start over in order to make it work well.  Nothing in WSPR changed, of course, and the transmission is still the transmission, it's the management of the transmissions surrounding it that were a throw-away.

The good news is I am about 90% there, so keep your eyes peeled for a new release.

I need to ask someone I know who does IP law about the licensing for this project. The original was licensed under GNU GPL, and I have a fundamental distaste for GPL. I release all of my projects under the MIT license; I may have to release the new code in a new repo—if I do, I will announce it appropriately here.

lbussy, 2025-03-28:
I am re-opening this as a tracker so there will be a notification when I do release.

lbussy, 2025-04-06:
I pinned a "use a current version" note here and in the latest release.  I don't think that's too much to ask, and even in 2.0, I will not remove the dependency on glibc.

closed 2025-04-06 by lbussy