WsprryPi/WsprryPi #101: Shutdown Watch Error Log Growing

AA7US, 2025-02-08:
Hi Lee,

I'm seeing an error (listed below) being continuously captured in the file: /var/log/wspr/shutdown_watch.error.log

The log file size keeps growing. Is this something I should be concerned about?

I'm not sure what version I'm running (or how to find out).

Thank you,

John L.

`lgpio.error: 'GPIO busy'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 411, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=35, name='GPIO19', names=frozenset({19, 'BOARD35', '19', 'GPIO19', 'WPI24', 'J8:35', 'BCM19'}), pull='', row=18, col=1, interfaces=frozenset({'', 'pwm', 'spi', 'pcm', 'gpio', 'dpi'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/shutdown_watch.py", line 78, in <module>
    main()
  File "/usr/local/bin/shutdown_watch.py", line 39, in main
    stop_button = Button(STOP_PIN) # defines the button as an object and chooses GPIO pin
                  ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 412, in __init__
    super().__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 417, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 162, in __init__
    super().__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 243, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 79, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 553, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 413, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))`

lbussy, 2025-02-11:
Yep, I broke it. :)

You can disable the shutdown_watch service for now.  Or, if you need a fix, I can see if I can do that RQ.

AA7US, 2025-02-12:
Thanks Lee. I do use the shutdown button, but am in no hurry for the fix. The shutdown button comes in handy when summer lightning storms approach and I need to get the Pi powered down and antenna disconnected quickly. The monsoon season doesn’t start here until late summer.

lbussy, 2025-02-12:
Here's a fix for you (I think - it's not going to break it worse either way):

```
sudo systemctl stop shutdown_watch
sudo systemctl disable shutdown_watch
```

Then. take [this file](https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/config/executables/shutdown_watch.py) and replace the contents of `/use/local/bin/shutdown_watch.py` with it.

Then replace the contents of `/lib/systemd/system/shutdown_watch.service` with this:

```
# Created for WsprryPi version 1.2.1-config+2f2caaa
# Copyright (C) 2023-2025 Lee C. Bussy (@LBussy)

[Unit]
Description=Wsprry Pi daemon for: shutdown_watch
Documentation=https://github.com/lbussy/WsprryPi/discussions
After=multi-user.target

[Service]
Type=simple
Restart=on-failure
RestartSec=5
User=root
Group=root
ExecStart=/usr/bin/python3 /usr/local/bin/shutdown_watch.py -w -D
SyslogIdentifier=shutdown_watch
StandardOutput=append:/var/log/wsprrypi/shutdown_watch_log
StandardError=append:/var/log/wsprrypi/shutdown_watch_error

[Install]
WantedBy=multi-user.target
```

Then, you can delete the old logs.  Then:

```
sudo systemctl enable shutdown_watch
sudo systemctl start shutdown_watch
```

That should get ya going.

AA7US, 2025-02-13:
That seems to have done the trick perfectly. One small item of note is that on my Pi OS Bookworm system, the "shutdown_watch.service" file is located in "/etc/systemd/system/" and not the "/lib/systemd/system/".

Thank you much!

lbussy, 2025-02-13:
Good deal.  It's fixed in the newer versions but they are not ready to release yet.  This will get you going for now. The log will not rotate, but there shouldn't be enough logging to worry unless you use the button every 3@ seconds.

lbussy, 2025-03-28:
I am re-opening this as a tracker so there will be a notification when I do release.

PR #132 (2025-04-06): Bugfixes 1.2.2
Closes #101 
Closes #68
closed 2025-04-06 by lbussy