WsprryPi/WsprryPi #111: Strange installation and 'Unable to retrieve data" error

gjmkoper, 2025-02-19:
I have an RPiZero and installed the latest os, Release date: November 19th 2024, System: 32-bit, Kernel version: 6.6, Debian version: 12 (bookworm).
After upgrading I installed in the recommended fashion: "curl -L installwspr.aa0nt.net | sudo bash"
First time rpi crashed during installation after TAPR-button question.
Second time it passed.
But, the when the html screen opens, the message "Unable to retrieve data" appears quickly.

lbussy, 2025-02-27:
I will give this excuse ONE LAST TIME:  I am not getting emails about these issues. My apologies.  Here is proof I changed my setting!

[image]

Now that THAT is over ... :)

Yes, this must run with sudo. I'm not sure how you are running this based on your report.  If you ran the installer, the executable should run under systemd with root privs; you would not have to run it intentionally.  You mentioned the web page, so it must be installed.

Check these things:

`sudo systemctl status wspr`

... please just paste in here what it says.

Also, share the logs located in: `/var/log/wspr/`, both `wspr.error.log` and `wspr.transmit.log`.

gjmkoper, 2025-02-27:
Thank you for your reply. It is some time since I posted this so I have to look this up again. I am now on holidays but when I am back I surely will give it another try and report to you on the result.

Ger

lbussy, 2025-02-27:
Yeah, sorry about that - hopefully, I really have the notifications fixed now.

Enjoy your vacation!

lbussy, 2025-03-18:
I am going to close this as resolved.  If you still have the issue, you can reply, and we can see what's what.

closed 2025-03-18 by lbussy

gjmkoper, 2025-03-26:
Dear all

Well, I bought a brandnew RPi Zero (no wifi) and tried again three times. Each time the installation crashed on another instant. Finally - I do not remember how - I got it installed but it does not run properly. Same problems as initially reported. 

I leave it by this. I found a very old version with only command line interface but that works just fine. Thanks for your attempt to make a more modern version.

Ger

lbussy, 2025-03-26:
I would still need the logs I asked for up above to be able to help.

gjmkoper, 2025-03-26:
OK, because you insist:

It is an RPi Zeo version 1.3, no wifi but a usb to internet connection.
After OS installation headless: no monitor, mouse or keyboard.
16 GB sd-card

cat /etc/os-release gives
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

sudo systemctl status wspr
● wspr.service - wspr daemon for: wspr
     Loaded: loaded (/etc/systemd/system/wspr.service; enabled; preset: enabled)
     Active: active (running) since Wed 2025-03-26 22:11:50 CET; 513ms ago
       Docs: https://github.com/lbussy/WsprryPi/discussions
   Main PID: 566 (wspr)
      Tasks: 1 (limit: 397)
        CPU: 226ms
     CGroup: /system.slice/wspr.service
             └─566 /usr/local/bin/wspr -D -i /usr/local/etc/wspr.ini

Mar 26 22:11:50 wsprrypi2 systemd[1]: Started wspr.service - wspr daemon for: wspr.

Also, share the logs located in: /var/log/wspr/, both wspr.error.log 

2025-03-26 21:07:45 GMT	2025-03-26 21:07:45 GMT	2025-03-26 21:07:51 GMT	2025-03-26 21:07:51 GMT	2025-03-26 21:07:59 GMT	2025-03-26 21:07:59 GMT	2025-03-26 21:08:12 GMT	2025-03-26 21:08:12 GMT	2025-03-26 21:08:18 GMT	2025-03-26 21:08:18 GMT	2025-03-26 21:08:24 GMT	2025-03-26 21:08:24 GMT	2025-03-26 21:08:30 GMT	2025-03-26 21:08:30 GMT	2025-03-26 21:08:36 GMT	2025-03-26 21:08:36 GMT	2025-03-26 21:08:42 GMT	2025-03-26 21:08:42 GMT	2025-03-26 21:08:48 GMT	2025-03-26 21:08:48 GMT	2025-03-26 21:08:53 GMT	2025-03-26 21:08:53 GMT	2025-03-26 21:08:59 GMT	2025-03-26 21:08:59 GMT	2025-03-26 21:09:05 GMT	2025-03-26 21:09:05 GMT	2025-03-26 21:09:10 GMT	2025-03-26 21:09:10 GMT	2025-03-26 21:09:16 GMT	2025-03-26 21:09:16 GMT	2025-03-26 21:09:45 GMT	2025-03-26 21:09:45 GMT	2025-03-26 21:09:51 GMT	2025-03-26 21:09:51 GMT	2025-03-26 21:09:57 GMT	2025-03-26 21:09:57 GMT	2025-03-26 21:10:04 GMT	2025-03-26 21:10:04 GMT	2025-03-26 21:10:09 GMT	2025-03-26 21:10:09 GMT	2025-03-26 21:10:15 GMT	2025-03-26 21:10:15 GMT	2025-03-26 21:10:21 GMT	2025-03-26 21:10:21 GMT	2025-03-26 21:10:27 GMT	2025-03-26 21:10:27 GMT	2025-03-26 21:10:33 GMT	2025-03-26 21:10:33 GMT	2025-03-26 21:11:00 GMT	2025-03-26 21:11:00 GMT	2025-03-26 21:11:05 GMT	2025-03-26 21:11:05 GMT	2025-03-26 21:11:11 GMT	2025-03-26 21:11:11 GMT	2025-03-26 21:11:16 GMT	2025-03-26 21:11:16 GMT	2025-03-26 21:11:22 GMT	2025-03-26 21:11:22 GMT	2025-03-26 21:11:27 GMT	2025-03-26 21:11:27 GMT	2025-03-26 21:11:33 GMT	2025-03-26 21:11:33 GMT	2025-03-26 21:11:39 GMT	2025-03-26 21:11:39 GMT	2025-03-26 21:11:44 GMT	2025-03-26 21:11:44 GMT	2025-03-26 21:11:50 GMT	2025-03-26 21:11:50 GMT	2025-03-26 21:11:56 GMT	2025-03-26 21:11:56 GMT	2025-03-26 21:12:01 GMT	2025-03-26 21:12:01 GMT	2025-03-26 21:12:07 GMT	2025-03-26 21:12:07 GMT	2025-03-26 21:12:12 GMT	2025-03-26 21:12:12 GMT	2025-03-26 21:12:18 GMT	2025-03-26 21:12:18 GMT	2025-03-26 21:12:23 GMT	2025-03-26 21:12:23 GMT	2025-03-26 21:12:29 GMT	2025-03-26 21:12:29 GMT	2025-03-26 21:12:34 GMT	2025-03-26 21:12:34 GMT	2025-03-26 21:12:40 GMT	2025-03-26 21:12:40 GMT	

and wspr.transmit.log

2025-03-26 21:07:45 GMT	Wsprry Pi v1.2.1 (main).
2025-03-26 21:07:45 GMT	Running on: Raspberry Pi 1 or Zero Model (BCM2835).
2025-03-26 21:07:45 GMT	Config loaded from: /usr/local/etc/wspr.ini
2025-03-26 21:07:45 GMT	Transmit Enabled: false
2025-03-26 21:07:45 GMT	Call Sign:
2025-03-26 21:07:45 GMT	Grid Square:
2025-03-26 21:07:45 GMT	Transmit Power: 0
2025-03-26 21:07:45 GMT	Frequencies:
2025-03-26 21:07:45 GMT	PPM Offset: 0.000000
2025-03-26 21:07:45 GMT	Do not use NTP sync: true
2025-03-26 21:07:45 GMT	Check NTP Each Run (default): false
2025-03-26 21:07:45 GMT	Use Frequency Randomization: false
2025-03-26 21:07:45 GMT	Power Level: 7
2025-03-26 21:07:45 GMT	Use LED: false
Error: must specify callsign, gridsquare, dBm, and at least one frequency.
Try: wspr --help
2025-03-26 21:07:50 GMT	Wsprry Pi v1.2.1 (main).
2025-03-26 21:07:51 GMT	Running on: Raspberry Pi 1 or Zero Model (BCM2835).
2025-03-26 21:07:51 GMT	Config loaded from: /usr/local/etc/wspr.ini
2025-03-26 21:07:51 GMT	Transmit Enabled: false
2025-03-26 21:07:51 GMT	Call Sign:
2025-03-26 21:07:51 GMT	Grid Square:
2025-03-26 21:07:51 GMT	Transmit Power: 0
2025-03-26 21:07:51 GMT	Frequencies:
2025-03-26 21:07:51 GMT	PPM Offset: 0.000000
2025-03-26 21:07:51 GMT	Do not use NTP sync: true
2025-03-26 21:07:51 GMT	Check NTP Each Run (default): false
2025-03-26 21:07:51 GMT	Use Frequency Randomization: false
2025-03-26 21:07:51 GMT	Power Level: 7
2025-03-26 21:07:51 GMT	Use LED: false
Error: must specify callsign, gridsquare, dBm, and at least one frequency.
Try: wspr --help
2025-03-26 21:07:59 GMT	Wsprry Pi v1.2.1 (main).
[... 710 lines cut ...]
2025-03-26 21:13:01 GMT	Frequencies:
2025-03-26 21:13:01 GMT	PPM Offset: 0.000000
2025-03-26 21:13:02 GMT	Do not use NTP sync: true
2025-03-26 21:13:02 GMT	Check NTP Each Run (default): false
2025-03-26 21:13:02 GMT	Use Frequency Randomization: false
2025-03-26 21:13:02 GMT	Power Level: 7
2025-03-26 21:13:02 GMT	Use LED: false
Error: must specify callsign, gridsquare, dBm, and at least one frequency.
Try: wspr --help
2025-03-26 21:13:07 GMT	Wsprry Pi v1.2.1 (main).
2025-03-26 21:13:07 GMT	Running on: Raspberry Pi 1 or Zero Model (BCM2835).
2025-03-26 21:13:07 GMT	Config loaded from: /usr/local/etc/wspr.ini
[... cut ...]
2025-03-26 21:13:07 GMT	Grid Square:
2025-03-26 21:13:07 GMT	Transmit Power: 0
2025-03-26 21:13:07 GMT	Frequencies:
2025-03-26 21:13:07 GMT	PPM Offset: 0.000000
2025-03-26 21:13:07 GMT	Do not use NTP sync: true
2025-03-26 21:13:07 GMT	Check NTP Each Run (default): false
2025-03-26 21:13:07 GMT	Use Frequency Randomization: false
2025-03-26 21:13:07 GMT	Power Level: 7
2025-03-26 21:13:07 GMT	Use LED: false
Error: must specify callsign, gridsquare, dBm, and at least one frequency.
Try: wspr --help

lbussy, 2025-03-26:
> OK, because you insist:

I'm trying to help, I do not insist.

Your logs indicate that the INI file located at `/usr/local/etc/wspr.ini` is not complete.  It should look like this:

```
[Control]
Transmit = False ; Bool

[Common]
Call Sign = NXXX ; String of 7 or less (I think)
Grid Square = ZZ99 ; Four characters
TX Power = 20 ; Power in dBm Always a relatively low integer
Frequency = 20m ; This can be an integer with "m" on the end, an integer as in '450000000', or an exponential format like '780e3'

[Extended]
PPM = 0.0 ; Double
Self Cal = True ; Bool
Offset = False; Bool
Use LED = False ; Bool
Power Level = 7 ; 0-7, 7 is max
```

Replace the call sign, of course.  Then try again and see what the transmit log says.

gjmkoper, 2025-03-27:
Dear Lee

help appreciated, but it becomes a chore. Now the After closing down yesterday, I cannot get it to start up properly anymore. After connecting the monitor it seems that after starting the daemons the system becomes unreachable. 

As said before, I have other things to do now. After some time - not clear how much - I will try again.

Ger

lbussy, 2025-03-27:
I am working to release v2 soon; it is a complete rewrite vs trying to keep this legacy code running.  Keep an eye out for that.