WsprryPi/WsprryPi #135: Documentation Updates Needed

AA7US, 2025-04-09:
Hi Lee,

I've successfully updated to WsprryPi v1.2.2 and followed your instructions about deleting old files. Everything seems to be working properly on my 3B+ so far. Thank you for the continued development and the improvements. It's very much appreciated.

I noticed that you've changed the main executable name from "wspr" to "wsprrypi" so that makes the command line operations as detailed in the docs outdated. My apologies if you already have this on your to-do list.

73,

John L.
AA7US

lbussy, 2025-04-09:
It is on the list for sure. I'm in a strange in-between area where I wanted to get a couple of fixes out, but I wasn't really ready to do the rest of the work. The next version is a major upgrade, for me anyway. If I'm doing things right, you should not see too many changes.

I've been experimenting a lot with type two and three messages, for those folks who want to explore using something other than the vanilla type one.

Been doing a fair bit of travel here lately, so that slowed me down. It feels like I can see the finish line, but it's backing up as fast as running towards it.

Such as a project like this, I guess.

lbussy, 2025-05-04:
Beta available (docs still not updated).  First, cleanup:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/old_version_cleanup.sh  | sudo bash
```

then:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/install.sh  | sudo bash
```

closed 2025-05-07 by lbussy