WsprryPi/WsprryPi #141: Version 2.0 Beta Failure at Launch

AA7US, 2025-05-04:
Greetings Lee,

Something is amiss with v2.0 beta installed on my Pi 3B+ because I'm getting a failure of wsprrypi to launch.

Here's a screenshot of the systemctl status:

[image]

It looks like wsprrypi might be looking for /usr/local/etc/wspr.ini at launch and not finding it.  Instead I see /usr/local/etc/wsprrypi.ini.

[image]

I'm also unable to save my settings on the web UI.

[image]

I did run both the v2.0 beta cleanup and install commands and rebooted (just for good measure). I ran the cleanup and install commands a second time with no change. No errors were noted.

BTW, the new web GUI looks GREAT!

73,

AA7US

lbussy, 2025-05-06:
This should be fixed, I can't imagine what I did, but the new Beta is moments away.

AA7US, 2025-05-07:
Unfortunately, even after successfully installing the updated 2.0 beta, my WsprryPi installation was STILL looking for wspr.ini instead of wsprrypi.ini and throwing the launch error. I edited /etc/systemd/wsprrypi.service manually as suggested which solved the problem.

Interestingly, my wsprrypi.service file says it was created for version 1.2.2, so maybe it's not being deleted and/or replaced by the cleanup and install scripts?

AA7US, 2025-05-07:
> See if there's an install log in your home directory. It might give a clue.

There's no install log in my home directory for the 2.0 beta.

Now there WAS an install log from one of the previous versions (I don't know which), but I cleaned out my home directory and deleted it prior to installing the 2.0 beta.

lbussy, 2025-05-07:
Found it.

I moved the old cleanup script into the installer.  Check.
I forgot to add the function to the installation orchestration.  Doh.

I'll mash this into the Beta 2 here as soon as I test it.

lbussy, 2025-05-07:
I do see the log when I execute install/uninstall:

```bash
pi@wspr4:~/WsprryPi $ ls -al ~/install.log
-rw-r--r-- 1 pi pi 12086 May  7 06:59 /home/pi/install.log
```

Not sure why you are not seeing it.

lbussy, 2025-05-07:
Okay, this is fixed, I'll pull into the Beta here shortly.  APPARENTLY, I need to call the function that cleans up the old versions to work.  :)

commit 7b1225c2 (2025-05-07): Fix install logic (#147)

* Fix uninstall logic

* Remove log target on uninstall

* Fix version

* Address skipped functions

* Add curl-based execution notes

Closes #141
commit be5b9424 (2025-05-07): Fix install logic (#147)

* Fix uninstall logic

* Remove log target on uninstall

* Fix version

* Address skipped functions

* Add curl-based execution notes

Closes #141
closed 2025-05-07 by lbussy