WsprryPi/WsprryPi #146: Error 404 Path to Beta 2.0

AA7US, 2025-05-06:
I'm getting an "error: 404" when trying to run the beta 2.0 install script using the path shown in Releases...

pi@WSPRPi:~ $ curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_Beta.2/scripts/install.sh | sudo bash

curl: (22) The requested URL returned error: 404

lbussy, 2025-05-06:
I just copied the URL again:

[https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_Beta/scripts/install.sh](https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_Beta/scripts/install.sh)

I see the issue - there's a `.2` in the URL I posted, which is the tag, not the branch.

I'll update the release notes.

AA7US, 2025-05-07:
That fixed it.

Thank you!

closed 2025-05-07 by AA7US