WsprryPi/WsprryPi #150: Continuous Errors Being Posted in Shutdown Watch Log

AA7US, 2025-05-08:
Running WsprryPi version "2.0_Beta.3-2-0-Beta (2-0-Beta)" I'm seeing "/var/log/wsprrypi/shutdown_watch.error.log" continuously growing with the following errors (there are no time / date stamps):

```
/usr/bin/python3: can't open file '/usr/local/bin/shutdown_watch.py': [Errno 2] No such file or directory
/usr/bin/python3: can't open file '/usr/local/bin/shutdown_watch.py': [Errno 2] No such file or directory
/usr/bin/python3: can't open file '/usr/local/bin/shutdown_watch.py': [Errno 2] No such file or directory
```

And obviously there's no "shutdown_watch.py" located in the "/usr/local/bin" directory. :)

John L.
AA7US

lbussy, 2025-05-08:
And see if there's a service running named "shutdown_watch". If so, stop it.

AA7US, 2025-05-08:
I rebooted immediately after running the beta install the last time and have rebooted again (several times actually) but the errors persisted.

As requested I checked to see if a "shutdown_watch" service was running (yes):

```
pi@WSPRPi:/var/log/wsprrypi $ sudo systemctl status shutdown_watch
● shutdown_watch.service - wsprrypi daemon for: shutdown_watch
     Loaded: loaded (/etc/systemd/system/shutdown_watch.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2025-05-07 17:53:42 MST; 1s ago
       Docs: https://github.com/lbussy/WsprryPi/discussions
    Process: 932 ExecStart=/usr/bin/python3 /usr/local/bin/shutdown_watch.py (code=exited, status=2)
   Main PID: 932 (code=exited, status=2)
        CPU: 109ms
```

I ran a stop command and rechecked the status:

```
pi@WSPRPi:/var/log/wsprrypi $ sudo systemctl status shutdown_watch
○ shutdown_watch.service - wsprrypi daemon for: shutdown_watch
     Loaded: loaded (/etc/systemd/system/shutdown_watch.service; enabled; preset: enabled)
     Active: inactive (dead) (Result: exit-code) since Wed 2025-05-07 17:54:52 MST; 2s ago
   Duration: 110ms
       Docs: https://github.com/lbussy/WsprryPi/discussions
    Process: 949 ExecStart=/usr/bin/python3 /usr/local/bin/shutdown_watch.py (code=exited, status=2)
   Main PID: 949 (code=exited, status=2)
        CPU: 109ms

May 07 17:54:52 WSPRPi systemd[1]: Stopped shutdown_watch.service - wsprrypi daemon for: shutdown_watch.
```

The errors have now stopped being posting to "shutdown_watch.error.log".

lbussy, 2025-05-08:
Now disable it to make sure it does not come back on reboot. I'll look into why this was missed on the cleanup.

lbussy, 2025-05-08:
You can, yes.   The shutdown button monitoring is in the Wsprry Pi executable now.

lbussy, 2025-05-08:
Okay, this is fixed in Beta.3 now.  If you have removed it, there's no need to do anything else.

closed 2025-05-08 by lbussy

lbussy, 2025-05-08:
Not sure if you get notifications (or are tired of the spam):

https://github.com/lbussy/WsprryPi/releases/tag/2.0_Beta.4

I think this has all bugs fixed, and I have an INI merge on upgrade in there.

AA7US, 2025-05-08:
> Not sure if you get notifications (or are tired of the spam):

Yep. I'm receiving email notifications for Issues, Releases and Discussions.

I'll keep testing.

Thanks!