WsprryPi/WsprryPi #153: Transmit Log Name Change in 2.0 Beta 4

AA7US, 2025-05-08:
I don't know if this is intentional, but after installing 2.0 Beta 4, the transmit log is now named "wsprrypi_log" (note the underscore) and not "wsprrypi.transmit.log" like it was with previous versions.

lbussy, 2025-05-08:
> AA7US created an issue [(lbussy/WsprryPi#153)](https://github.com/lbussy/WsprryPi/issues/153)

Well, it's been that way for a while. I think the bug in the installer that caused you all the other grief is the culprit for not seeing it.

It's a single log, no second error log. And see the log viewer up top there in the nav bar?

AA7US, 2025-05-08:
You're probably right about that.

I did see "WSPR Logs" in the web GUI menu bar yesterday and tried it, but I only got a black screen where the log should be and decided to wait to troubleshoot that until later on since we had other issues going on. It works now so everything seems to be functioning as designed.

I'll close this.

Sorry for the false alarm.

73,

John L.
AA7US

closed 2025-05-08 by AA7US