WsprryPi/WsprryPi #154: Test Tone from Command Line Not Working

AA7US, 2025-05-08:
Running 2.0 Beta 4

I don't know if I'm doing something wrong, but I can't get either "-t" or "--test-tone" to work anymore.

```
pi@WSPRPi:~ $ sudo wsprrypi -t 10140200
[INFO ] WsprryPi version 2.0_Beta.4-2-0-Beta (2-0-Beta).
[INFO ] Running on: Raspberry Pi 3 Model B Plus Rev 1.4.
[INFO ] Process PID: 881
[INFO ] WSPR loop running.
[ERROR] Empty or invalid frequency list; disabling transmission.
[INFO ] A test tone will be generated at 10.140200 MHz .
```

```
pi@WSPRPi:~ $ sudo wsprrypi --test-tone 10140200
[INFO ] WsprryPi version 2.0_Beta.4-2-0-Beta (2-0-Beta).
[INFO ] Running on: Raspberry Pi 3 Model B Plus Rev 1.4.
[INFO ] Process PID: 877
[INFO ] WSPR loop running.
[ERROR] Empty or invalid frequency list; disabling transmission.
[INFO ] A test tone will be generated at 10.140200 MHz .
```

Note the entry: "[ERROR] Empty or invalid frequency list; disabling transmission."

No signal is generated and I have to hit ctrl-c to get the bash command prompt back.

lbussy, 2025-05-08:
It's like whack-a-mole. :)

Hey, the important part works!

AA7US, 2025-05-08:
Indeed! WsprryPi is now a WSPR beaconer’s dream come true and it’s 99% there. You’re doing amazing work.

Here’s a crazy idea (a feature request), how about eventually adding a switch to the web GUI to generate a test tone instead of having to use the CLI? I use the test tone often to adjust my antenna matching unit.

lbussy, 2025-05-08:
[image]

lbussy, 2025-05-08:
Fixed in Devel.  If you want to test:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/install.sh | sudo bash
```
(same for both bugs)

AA7US, 2025-05-08:
Tested here and it's now working perfectly.

Thanks Lee!

PR #156 (2025-05-08): Fix test tone and CLI Option
Closes #154 & #155
closed 2025-05-08 by lbussy