WsprryPi/WsprryPi #155: CLI Short Option for Offset Not Working

AA7US, 2025-05-08:
Running v2.0 Beta 4 from the command line.

The short option "-o" throws an "invalid option" error, but "--offset" works just fine.

```
pi@WSPRPi:~ $ sudo wsprrypi -o AA7US DM41 27 30m
wsprrypi: invalid option -- 'o'

WsprryPi version 2.0_Beta.4-2-0-Beta (2-0-Beta).

Usage:
  (sudo) wsprrypi [options] callsign gridsquare transmit_power frequency <f2> <f3> ...
    OR
  (sudo) wsprrypi --test-tone {frequency}

Options:
  -h, --help
    Display this help message.
  -v, --version
    Show the WsprryPi version.
  -i, --ini-file <file>
    Load parameters from an INI file. Provide the path and filename.

See the documentation for a complete list of available options.
```

```
pi@WSPRPi:~ $ sudo wsprrypi --offset AA7US DM41 27 30m
[INFO ] WsprryPi version 2.0_Beta.4-2-0-Beta (2-0-Beta).
[INFO ] Running on: Raspberry Pi 3 Model B Plus Rev 1.4.
[INFO ] Process PID: 1052
[INFO ] WSPR loop running.
[INFO ] WSPR packet payload:
[INFO ] - Callsign: AA7US
[INFO ] - Locator: DM41
[INFO ] - Power: 27 dBm
[INFO ] Requested TX frequencies:
[INFO ] - 10.140200 MHz
[INFO ] TX will stop after: 1 iteration(s) of the frequency list.
[INFO ] A random offset will be added to all transmissions.
[INFO ] DMA setup complete, waiting for next transmission window.
```

lbussy, 2025-05-08:
Fixed in Devel.  If you want to test:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/install.sh | sudo bash
```
(same for both bugs)

AA7US, 2025-05-08:
Tested "-o" and it's now working perfectly.

Thank you!

closed 2025-05-08 by lbussy