WsprryPi/WsprryPi #157: Possible Errant Control Character in Log File

AA7US, 2025-05-08:
I apologize for opening yet another issue. Hopefully this will be it for awhile.

I'm still running the last dev version of 2.0 Beta 4 you provided.

In wsprrypi's unified log there appears to be a hidden null character (control-@) that is causing my terminal program to halt display of the log when encountered. The hidden character is located immediately after the Raspberry Pi's model number (see below). I'm only able to force my terminal to get past it (the null character) requesting it display all characters (including hidden).

```
2025-05-08 21:59:09.373 UTC^I[INFO ] WsprryPi version 2.0_Beta.2-2-0-devel+0c60b4e (2-0-devel).$
2025-05-08 21:59:09.374 UTC^I[INFO ] Running on: Raspberry Pi 3 Model B Plus Rev 1.4^@.$
2025-05-08 21:59:09.374 UTC^I[INFO ] Process PID: 900$
2025-05-08 21:59:09.375 UTC^I[INFO ] Web server started on port: 31415$
2025-05-08 21:59:09.376 UTC^I[INFO ] WebSocketServer started on port: 31416$
2025-05-08 21:59:09.377 UTC^I[INFO ] WSPR loop running.$
2025-05-08 21:59:09.381 UTC^I[INFO ] WSPR packet payload:$

```
Does this null character have to be there in the log?

On the bright side, this issue doesn't seem to bother displaying the log accessed via the web GUI.

lbussy, 2025-05-08:
So it's your problem, not mine?  ;)

I'm sure I can find it.

lbussy, 2025-05-08:
It looks like the tab character is not displaying correctly either - should be a tab between UTC and the open brace.

What terminal application are you using?  Some Windows-based one?

AA7US, 2025-05-08:
> So it's your problem, not mine? ;)

Definitely! :)

I'm using WebSSH (https://github.com/isontheline/pro.webssh.net), a cross platform terminal for Mac and iOS.

Thanks for looking into it Lee. Much appreciated.

AA7US, 2025-05-09:
> It looks like the tab character is not displaying correctly either - should be a tab between UTC and the open brace.

To clarify the situation, everything in the log file is displaying correctly, but it's only that hidden control-@ being a problem. The example that I posted yesterday was created with the command "cat -A /var/log/wsprrypi/wsprrypi_log". The "-A" option forcing the display of all characters, even normally hidden tab control characters.

Bottom line: There's nothing wrong with the tabs, it's just that pesky control-@ character creating a problem.

Here's a sample output using "tail -n 5 /var/log/wsprrypi/wsprrypi_log":

```
pi@WSPRPi:~ $ tail -n 5 /var/log/wsprrypi/wsprrypi_log
2025-05-09 13:38:01.000 UTC     [INFO ] Transmission started (frequency = 10.140200 MHz).
2025-05-09 13:39:51.629 UTC     [INFO ] Transmission complete (110.628 secs).
2025-05-09 13:39:51.660 UTC     [INFO ] DMA setup complete, waiting for next transmission window.
2025-05-09 13:40:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
2025-05-09 13:42:01.000 UTC     [INFO ] Skipping transmission (frequency = 0.0).
```

All the tabs after UTC look good.

Thanks!

AA7US, 2025-05-09:
That worked beautifully! The null character is gone and log lists without hanging.

Thank you Lee!

73,

John L.
AA7US

lbussy, 2025-05-09:
You are in a dead-end branch right now; I'll get this promoted to devel and then to Beta in a bit.  You should be able to run the installer again when I get there to get the latest.

I appreciate all your testing!  It's a huge help.

closed 2025-05-09 by lbussy

lbussy, 2025-05-09:
This is in Beta.5, currently unpublished.  If you want to be at the tip of HEAD you can apply this over what you have:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_Beta/scripts/install.sh | sudo bash
```

AA7US, 2025-05-09:
> If you want to be at the tip of HEAD you can apply this over what you have:

Absolutely!

Beta 5 is installed and running properly.

Thank you,

John L.
AA7US

commit d515d398 (2025-05-09): Fixes #157 (null terminated string in logs)
commit dea46212 (2025-05-09): Fixes #157 (null terminated string in logs) (#159)