WsprryPi/WsprryPi #160: Add Test Tone

lbussy, 2025-05-10:
Add a test tone capability.

[image]

It's currently just a button while I think about if I like it there or not.

lbussy, 2025-05-11:
@AA7US, when you use the test tone, do you expect/want the same PPM adjustment that WSPR uses?  In other words, would you set the configuration with NTP or PPM and then generate a tone at whatever frequency you want?

Would you want it to generate the tone at whatever is set up in Frequencies?

AA7US, 2025-05-11:
@lbussy,

Re: Frequency - I think it'd be quite convenient to have the test tone function use whatever frequency has been entered in the "Frequencies" field (as opposed to having to fill out another setup screen). Wait, can more than one frequency be entered in the Frequencies field? I ask because I only ever use one frequency as I use single band antennas. If more than one frequency can be entered, then obviously you'll have to figure out a way to deal with that. Maybe only use the first frequency entered?

Re: NTP and PPM - I currently don't need to use NTP or PPM adjustments so I'm agnostic as to whether they're included for generating a test tone. I can't imagine any harm in including them, especially if it makes setting this up easier for you.

FYI: I only use the test tone once after changing bands to adjust my antenna matching unit. The exact frequency isn't super critical. I normally just set the frequency to the middle of the WSPR passband.

I hope that helps.

AA7US, 2025-05-11:
When will the WsprryPi merch be available?

[image]

lbussy, 2025-05-11:
Thanks for the feedback - that's where my head was going with it.  I'm sure SOMEONE will tell me if you're wrong.

Wee what I did there?  :)

AA7US, 2025-05-11:
> See what I did there? :)

Yes. Very clever. ;)

lbussy, 2025-05-16:
@AA7US that, and more, is ready to test:

https://github.com/lbussy/WsprryPi/releases/tag/2.0_Beta.6

AA7US, 2025-05-16:
Hey Lee,

Something weird has happened...

Prior to beginning the install of v2.0 beta 6, I decided to pull up WsprryPi's web GUI and found the formatting all whacked (i.e.: the page wasn't rendered correctly). Thinking this might be an Apple Safari web browser problem, I tried again with the Opera browser, but encountered same problem with messed up formatting. I do know the web GUI was working fine initially after the last update, but since I haven't accessed it in a few days, I don't know when it decided to quite working.

I rebooted the Pi and encountered the same problem.

Then I decided to go ahead and install v2.0 beta 6 figuring that should clean up whatever was causing the web GUI problem and noticed a failure and warning in the script output before it returned to the command prompt:

```
[INFO ] Deleting local repository tree.
[✔] Complete: Delete local repository.
[✔] Complete: Adding ServerName directive.
[✔] Complete: Writing /etc/apache2/sites-available/wsprrypi.conf.
[✔] Complete: Disabling default site.
[✔] Complete: Enabling wsprrypi site.
[✘] Failed: Testing Apache configuration.
[WARN ] [install.sh:exec_command:4585] Command failed with status 1: apache2ctl configtest
```

The following are today's lines from /home/pi/wsprrypi.log:

```
2025-05-15 19:07:57 [INFO ] [piped_script:3288] Checking environment.
2025-05-15 19:07:58 [INFO ] [piped_script:3288] System: Raspbian GNU/Linux 12 (bookworm).
2025-05-15 19:07:58 [INFO ] [piped_script:3288] Running Wsprry Pi's install script, version 2.0_Beta.6.
2025-05-15 19:07:58 [INFO ] [piped_script:3288] Updating and managing required packages (this may take a few minutes).
2025-05-15 19:08:45 [INFO ] [piped_script:3288] APT package handling completed successfully.
2025-05-15 19:08:45 [INFO ] [piped_script:3288] Wsprry Pi installation beginning.
2025-05-15 19:09:24 [INFO ] [piped_script:3288] Cleaning up older services.
2025-05-15 19:09:29 [INFO ] [piped_script:3288] Cleaning up older files and directories.
2025-05-15 19:09:29 [INFO ] [piped_script:3288] Installing wsprrypi.
2025-05-15 19:09:30 [INFO ] [piped_script:3288] Installing 'wsprrypi' configuration.
2025-05-15 19:09:30 [INFO ] [piped_script:3288] Merged /usr/local/etc/wsprrypi.ini into new config.
2025-05-15 19:09:34 [INFO ] [piped_script:3288] Creating systemd service: wsprrypi.service.
2025-05-15 19:09:38 [INFO ] [piped_script:3288] Systemd service wsprrypi created.
2025-05-15 19:09:38 [INFO ] [piped_script:3288] Installing 'logrotate' configuration.
2025-05-15 19:09:38 [INFO ] [piped_script:3288] Installing web files to '/var/www/html/wsprrypi'.
2025-05-15 19:09:39 [INFO ] [piped_script:3288] Deleting local repository tree.
```

After installing v2.0 beta 6, the web GUI appears to be rendering correctly, but with obvious problems:

1. At the bottom of the screen the version info says "Loading version..."
2. The time shown in the upper right part of the screen isn't active (just dashes)
3. I can't save any entries that I try to make.

I poked around in Apache's log files and didn't see anything obvious, but admittedly I don't know what to look for.

Can you please advise what steps to try next?

Thanks!

AA7US, 2025-05-16:
I forgot to mention that after installing v2.0 beta 6 WsprryPi appears to be running fine using the last parameters retained in /usr/local/etc/wsprrypi.ini.

lbussy, 2025-05-16:
Minor note at 07:11 CST - I missed a line in the installer that is now fixed, so if you beat me to it, re-run.  All good now.

AA7US, 2025-05-16:
> Email all to me. My email is exactly what you would expect at ARRL.

Email sent.

lbussy, 2025-05-17:
Closing this one .... here's the last beta before it goes to RC.  No changes vs what you have, just experimenting with some release orchestration:

```bash
curl -fsSL https://raw.githubusercontent.com/lbussy/WsprryPi/refs/heads/2.0_devel/scripts/install.sh | sudo bash
```

closed 2025-05-17 by lbussy