WsprryPi/WsprryPi #166: PPM Offset Error in Web GUI

AA7US, 2025-05-17:
Running v2.0 beta 6

I entered "0" in the web GUI for "PPM Offset" field and clicked the "Save" button.

/usr/local/etc/wsprrypi.ini shows this choice correctly set:

```
[Extended]
; PPM: Frequency offset in parts per million (floating-point).
PPM = 0
; Use NTP: Calibrate tones via NTP
Use NTP = false
; Offset: Set to True to enable frequency offset correction, False to disable.
Offset = true
; Use LED: Set to True to enable LED usage, False to disable.
Use LED = true
; LED Pin as a GPIO/BCM designation, e.g. Pin 12/GPIO18/BCM18 = 18
LED Pin = 18
; Power Level: Output power level (integer from 0 to 7, where 7 is maximum).
Power Level = 7
```

If I reload the web GUI, the PPM Offset field is blank and indicates "Invalid". 

[image]

I can enter "0" again in PPM Offset to continue using the web GUI, but the problem reoccurs whenever I reload the page.

lbussy, 2025-05-17:
Got it, bug squarshed.

Apparently `$("#ppm.val").val(ppm);` is not the same as `$("#ppm").val(ppm);`.

In reality, it never saved the zero either, it saved the default, which is 0.0.  Had you changed it to any other value, it would have saved it as 0.

RC 1 will be out momentarily with a fix.

AA7US, 2025-05-17:
Glad you found it!

After running overnight, I just checked the web GUI again this morning and it still looks good.

AA7US, 2025-05-17:
FYI: Running RC1 now but was still experiencing the same problem with the PPM field. I tried a different browser and no problem. Huh? Cleared the cache on browser having the problem and it was cured.

lbussy, 2025-05-17:
Shift – refresh, always shift – refresh when you're looking for a new version of a webpage.

AA7US, 2025-05-17:
> Shift – refresh, always shift – refresh when you're looking for a new version of a webpage.

Thanks! If I ever knew you that could force refresh an individual website's data, I'd forgotten it long ago and have always resorted to cleaning the entire cache, which comes at a cost. With some help from AI, I just found out the keyboard shortcut when using Safari browser on Mac OS is Command-R (and the browser's developer mode has to be enabled first). Unfortunately there's no keyboard shortcut for either iPhone or iPad OS and it instead requires digging several layers deep into the OS settings menus.