WsprryPi/WsprryPi #173: Randomize - Logging or Logic Issue?

lbussy, 2025-06-05:
```text
2025-06-05 16:17:16.342 UTC [INFO ] Requested TX frequencies:
2025-06-05 16:17:16.342 UTC [INFO ] - 7.040100 MHz
2025-06-05 16:17:16.342 UTC [INFO ] A random offset will be added to all transmissions.
2025-06-05 16:17:16.376 UTC [INFO ] DMA setup complete, waiting for next transmission window.
2025-06-05 16:47:14.676 UTC [INFO ] Transmission started at 7.040100 MHz.
```

This implies the transmission is not randomized.  Either the logic is off, or the logging is.

AA7US, 2025-06-05:
I noticed this anomaly and have verified the transmit frequency IS being randomized when the option is selected, despite the log suggesting it’s not.

lbussy, 2025-06-05:
Well, I knew I heard laughter somewhere.

It was not as straightforward as I thought, but someone thought to put an optional messsage in the start_tx callbacks so I had a model to use.  It required a [change in the WSPR-Transmitter class](https://github.com/lbussy/WSPR-Transmitter/commit/863264bdb9e925a47d8b4f17ff61f52bf565fe4b) as well as in [WsprryPi](https://github.com/lbussy/WsprryPi/commit/e78edfc9ff7d774c0c40762da1f8bb086f62b515).

Seems to work now:

```text
2025-06-05 17:52:29.694 UTC [INFO ] WsprryPi version 2.0_Beta.6-randomize+7e07315 (randomize).
[...]
2025-06-05 17:52:29.806 UTC [INFO ] Requested TX frequencies:
2025-06-05 17:52:29.806 UTC [INFO ] - 3.570100 MHz
2025-06-05 17:52:29.806 UTC [INFO ] A random offset will be added to all transmissions.
[...]
2025-06-05 17:54:01.001 UTC [INFO ] Transmission started at 3.570064 MHz.
```

There's no test image for this, you'll just have to believe me. :)

lbussy, 2025-06-05:
Beta version with (proposed) fix:

https://github.com/lbussy/WsprryPi/releases/tag/2.0.1_Beta.1

closed 2025-06-08 by lbussy