WsprryPi/WsprryPi #174: Log Rotate Sometimes Fails

lbussy, 2025-06-05:
Log rotation sometimes fails, leaving strangely named or uncompressed output.

lbussy, 2025-06-05:
Beta version with (proposed) fix:

https://github.com/lbussy/WsprryPi/releases/tag/2.0.1_Beta.1

lbussy, 2025-06-12:
Hm ... well, that's visually worse, but like you said, there are logs. :)

The remaining two options are:

1. I do a crapton of work to harden this.
2. I use the hardening someone else wrote in `journald` and move the logs there

Moving to `journald` is technically a breaking change, but I doubt anyone built any of their own functionality around it this quickly.

AA7US, 2025-06-13:
Just for fun (and probably aggravation), I'm going to give the "copytruncate" directive a try in WsprryPi's logrotate.d configuration file. I'll let you know what happens.

lbussy, 2025-06-13:
I suspect it may be more consistent, at the loss of some data.   The logs rotate at midnight and the transmission, if enabled, kicks off a second later.

lbussy, 2025-06-13:
I went another way ... 

It struck me that if I lose lines because of the timing, I need a way to log then rotate.  So, [9746ffa](https://github.com/lbussy/WsprryPi/commit/9746ffa04e7cdf1aa92ceb05a96df5132ebd1f72) adds a size-based rotate vs time.  That means it will write the log, THEN check to see if we hit the size limit, then rotate.

I will revert some of the changes for #183 and rethink that class.  All that means is that I should be putting a stable beta out here shortly.

lbussy, 2025-06-13:
Beta.4 is there for you to try out. I backed out the new class that was giving us errors and crashes.

AA7US, 2025-06-13:
Installed and testing now.

Thanks!

lbussy, 2025-06-13:
At least we have gotten past where you come back and tell me "it didn't install!"  :)

AA7US, 2025-06-14:
With v2.0 beta 4 running overnight, the log file doesn’t appear to have rotated. Maybe that's expected after only one night? Listing journalctl shows the logrotate system did occur at midnight. I'll keep it running and see what it looks like tomorrow morning.

Also, I'm still seeing the following error in the wsprrypi_log:

```
878  terminate called after throwing an instance of 'std::runtime_error'
879    what():  MMapRegion: mmap failed: Cannot allocate memory
```

The last two of those errors occurred just under 7 hours apart.

lbussy, 2025-06-14:
Now that's weird - there are no changes to the DMA stuff in that version, which means this particular bug was preexisting.

On the rotate, that's expected.  

``` text
/var/log/wsprrypi/* {
    rotate 14
    size 10M
    compress
    delaycompress
    missingok
    notifempty
    sharedscripts
    copytruncate
    create 0644 root www-data
    postrotate
        if systemctl list-units --type=service | grep -q "wsprrypi.service"; then
            systemctl restart wsprrypi.service
        fi
    endscript
}
```

The `size 10M` means the log will rotate when it hits 10 megs.  My theory is the daemon will write, which puts it over 10M, and THEN the log rotates.  Since the daemon already wrote its stuff, there's no loss of data.

lbussy, 2025-06-14:
I am working on the DMA part now, so let's not worry too much about that.  As long as it restarted, I'm happy enough for now.

AA7US, 2025-06-14:
Ah ha! The 10M directive is likely the reason I see no log rotation. Currently my log is only 84K. It may take a very long time for the log to reach 10M.

Are you comfortable with me temporarily removing the 10M directive in v2.0beta 4 for more intensive testing of the log rotation?

AA7US, 2025-06-14:
Done!

lbussy, 2025-06-15:
Nice!  No lost lines in between?

AA7US, 2025-06-15:
> Nice! No lost lines in between?

As best I can tell, no lines were lost.

AA7US, 2025-06-16:
Update: No change from yesterday morning.

Everything seems fine, but strangely, no log rotation occurred at midnight. I think there should’ve been a rotation and a first log file compression. This is what I have as of now:

```
-rw-r--r-- 1 root www-data 133710 Jun 16 06:16 wsprrypi_log
-rw-r--r-- 1 root www-data 159045 Jun 15 00:00 wsprrypi_log.1
```

Journalctl indicates a log rotation on system did occur at midnight so I really don’t understand this behavior. I could understand it if the “size 10M” directive was still active, but it’s not.

Oh well, I'll let it keep running and see what happens overnight.

AA7US, 2025-06-17:
After another night without WsprryPi's logfile rotating, I asked my favorite AI bot for some suggestions on why this might be happening. Following its instructions to test what logrotate is doing in debug mode, I quickly discovered the global configuration for logrotate on my Raspberry Pi is set to WEEKLY rotations. Doh! Boy do i feel dumb.

I guess there’s nothing inherently wrong with having weekly rotations for WsprryPi’s log, but I'm going to add a “daily” directive to its configuration file because I want more opportunities to verify we don’t have any problems.

AA7US, 2025-06-17:
After adding the daily directive for WsprryPi's logfile then forcing a log rotation, it went from this:

```
-rw-r--r-- 1 root www-data 239001 Jun 17 05:54 wsprrypi_log
-rw-r--r-- 1 root www-data 159045 Jun 15 00:00 wsprrypi_log.1

```
to this:

```
-rw-r--r-- 1 root www-data   1436 Jun 17 06:22 wsprrypi_log
-rw-r--r-- 1 root www-data      0 Jun 17 06:21 wsprrypi_log.1
-rw-r--r-- 1 root www-data 240782 Jun 17 06:21 wsprrypi_log.1.1
-rw-r--r-- 1 root www-data  10814 Jun 15 00:00 wsprrypi_log.2.gz
```

Weird. WsprryPi was standing by waiting for the next time-slot and not transmitting when I forced the rotation.

AA7US, 2025-06-17:
Why do I feel like I'm obsessing over a few log files? Maybe i need to walk away. 🙂

lbussy, 2025-06-17:
I think the daily directive was on there before I changed it to size based. That makes sense.

lbussy, 2025-06-18:
I crammed this into devel and Beta.4 so re-running the installer should (re) apply it.

lbussy, 2025-06-22:
Beta.5 has this:

[https://github.com/lbussy/WsprryPi/releases/tag/2.0.1_Beta.5](https://github.com/lbussy/WsprryPi/releases/tag/2.0.1_Beta.5)

There's still some known weirdness when stopping/starting transmission via the web page.  I also dropped the hooks in for randomization per transmission, but I need to clean up the `WSPR-Transmitter` class before I wire it in.

AA7US, 2025-06-22:
v2.0.1 beta 5 installed smoothly.

> There's still some known weirdness when stopping/starting transmission via the web page.

Not sure if this is what you were referring to, but I just tried initiating a test tone from the web console (which I haven't tried in awhile) and it didn't work. I clicked on the "Start" button of the test tone several times and apparently crashed the program. WsprryPi kept restarting and crashing until I rebooted. Here's what was showing in the log at each crash (repeating):

```
2025-06-22 01:50:30.432 UTC     [INFO ] WsprryPi version 2.0.1_Beta.5-heads-2-0-1-Beta-5 (heads-2-0-1-Beta-5).
2025-06-22 01:50:30.438 UTC     [INFO ] Running on: Raspberry Pi 3 Model B Plus Rev 1.4.
2025-06-22 01:50:30.438 UTC     [INFO ] Process PID: 1036
2025-06-22 01:50:30.439 UTC     [INFO ] Web server started on port: 31415
2025-06-22 01:50:30.440 UTC     [INFO ] WebSocketServer started on port: 31416
2025-06-22 01:50:30.440 UTC     [INFO ] WSPR loop running.
2025-06-22 01:50:30.443 UTC     [INFO ] WSPR packet payload:
2025-06-22 01:50:30.443 UTC     [INFO ] - Callsign: AA7US
2025-06-22 01:50:30.443 UTC     [INFO ] - Locator: DM41
2025-06-22 01:50:30.443 UTC     [INFO ] - Power: 27 dBm
2025-06-22 01:50:30.444 UTC     [INFO ] Requested TX frequency: 10.140200 MHz
2025-06-22 01:50:30.444 UTC     [INFO ] A random offset will be added to all transmissions.
2025-06-22 01:50:31.484 UTC     [ERROR] Unhandled exception in main(): mem_alloc(): C‐style failed: Connection timed out
2025-06-22 01:50:31.484 UTC     [INFO ] WsprryPi exiting.
```

If I don't mess with the test tone, it seems to be running fine.

lbussy, 2025-06-22:
Might be related.   Disable transmissions and wait till it stops, then try test time.  May work better till I look at it tomorrow (I hope).

AA7US, 2025-06-22:
Okay great. I just checked in on how beta 5 ran overnight and it crashed continuously. 🙁

lbussy, 2025-06-22:
Okay re-run the installer and you will probably need to reboot to un-eff things. :)

AA7US, 2025-06-22:
> Okay re-run the installer and you will probably need to reboot to un-eff things. :)

Thank you!

Resuming testing now.

lbussy, 2025-06-25:
Gonna close this one out as fixed in Dev.

PR #184 (2025-06-13): Rotate logs
Intended to fix #174, move to size-based vs time-based rotation.