WsprryPi/WsprryPi #180: 2.0 Beta 2 Log Error

AA7US, 2025-06-06:
With v2.0 Beta 2, I'm seeing this new error in the wsprrypi_log after each WSPR transmission:

...snip...

2025-06-06 16:21:56.998 UTC     [INFO ] A random offset will be added to all transmissions.
2025-06-06 16:21:57.030 UTC     [INFO ] DMA setup complete, waiting for next transmission window.
2025-06-06 16:22:01.001 UTC     [INFO ] Transmission started at 10.140175 MHz.
2025-06-06 16:23:51.627 UTC     [INFO ] Transmission complete (110.627 secs).
terminate called after throwing an instance of 'std::system_error'
  what():  mbox_open(): cannot open /dev/vcio: Success

Afterwards, WsprryPi restarts.

lbussy, 2025-06-06:
I have been tracking that - I am playing with it in a dev branch.  I was NOT getting it between each transmission, but I was getting it after some form of reconfiguration (frequency skip, PPM update, etc.). Can you check to see if that matches on your side?

AA7US, 2025-06-06:
I think you're on the right track. Limited testing here shows no error when Wsprry Pi is configured to transmit a beacon every 2 minute cycle. As soon as I add a "0" or more to skip a cycle(s) (i.e.: 30m 0), the error is thrown and Wsprry Pi restarts.

lbussy, 2025-06-07:
Okay - fix is in [Beta.3](https://github.com/lbussy/WsprryPi/releases/tag/2.0.1_Beta.3).

lbussy, 2025-06-08:
Okay, this one is closed, #183 is the result. :)

Closing this one and I'll work on the next.

closed 2025-06-08 by lbussy