WsprryPi/WsprryPi #191: Proposal idea...

edegraaff, 2025-06-14:
I know there are some implementations to deliver a more exact location at the grid something that serve me well, and i know some collegues also would be happy to have the 4 digit grid as primary grid but support for the 6 position grid to. From what i read in the docu, the current implementation is the Type 1 wspr. 
I would like to be able to use the the type2 and or type 3 models to? 

https://www.researchgate.net/publication/377669485_HF_Frequency_Multi_Band_WSPR_Beacon_with_Energy_Storage_System_based_on_MPPT_Charge_Controller

Just a question/request 

Regards 
Eelco de PD1EG

edegraaff, 2025-06-15:
thumbs up

lbussy, 2026-04-02:
The Raspberry Pi 5 support will be via Si5351, which requires some abstraction of the GPIO-based transmission methods.  Those, in turn, allowed me to factor in (futore) support for Type 2/3 and other modes (QRSS, FSKCW, DFCW).

Making progress!

lbussy, 2026-04-07:
If anyone would like to test this at the command line (or at least limit INI updates to manual updates), let me know.

lbussy, 2026-04-08:
I'll get you some instructions here in a little bit.

lbussy, 2026-04-18:
Weird, when I click on that, I see a non-printing character at the end:

[image]

It should be:

```bash
curl -fsSL https://raw.githubusercontent.com/WsprryPi/WsprryPi/refs/heads/devel/scripts/install.sh | sudo bash
```

I have a new fix going in shortly for the GPIO/LPF control, but I have not changed this code (WSPR 2/3) in a bit.

edegraaff, 2026-04-18:
I am testing but.. i don;t know why i did not decode any wspr message witht he <pd1eg> and long grid notition. I will fall back to the old school model normal call and 4digit grid and see it this works well. 
hum .. it could be this ...
2026-04-18T18:41:34.009Z wsprrypi.service [INFO ] Requested WSPR dial frequency: 7.038600 MHz (GPIO25L)
2026-04-18T18:41:34.009Z wsprrypi.service [INFO ] A random offset will be added to all transmissions.
2026-04-18T18:41:34.034Z wsprrypi.service [INFO ] Waiting for next transmission window.
2026-04-18T18:42:01.001Z wsprrypi.service [INFO ] Started transmission: 7.040160 MHz (**GPIO25L)**.

lbussy, 2026-04-26:
I created a new Issue (#299) for your caching issue.  I am going to close this Issue (Type 2/3) because it has been delivered to the RC.   I'm losing the lead on some of this feedback, so when I push RC2 with a fix for another issue, I will start a new thread for any RC2 issues.  From there, I will split out any reports that need work into a separate thread.

I appreciate everyone's help!

closed 2026-04-26 by lbussy