WsprryPi/WsprryPi #209: wsprry service won't start: code=exited, status=203/EXEC

jdburke2k, 2025-07-06:
After installing 2.1.0 on pi zero w, a couple of times to get a clean completion of the install.sh script, the webpage comes back with no default values and broken functionality (for example, enabling the TX Led and Shutdown pin sliders and drop down boxes, refreshes the page and purges any other values. In general, Save does not seem to work).

Checking status of services shows enabled but not started.
Starting service shows error.
troubleshooting "ini" file and setting values manually did not seem to help. Noted no +X on the file.

[image]

jdburke2k, 2025-07-09:
Yes but not successfully...
1) The reboot/reinstall helped.
2) I was not able to connect with wifi after...but I think that is not a Wsprry issue but rather a pi zero w architecture problem.
3) in troubleshooting #2 above, I started to collect evidence that 
a) installing wsprry takes as least 2 reboots and I think 3. Install 1 breaks ssh and hangs during the last phases. I have to confirm that I can get a clean install on attempt 2...but there have been times that it took 3 (but I suspect that is a pi zero w thing not a wsprry thing)
b) I seem to have better luck on Bullseye headless rather than Bookworm headless but I am not ready to commit to that statement yet.
4) The attempt that is in progress now is "going well" - clean reinstall of raspi, and starting the 2 attempts at wsprry (which I think will work) and then fuss with wifi power and reattachment settings

Somewhere in there, a new version of wsprry was released, but I don't think that impacts the install script much

jdburke2k, 2025-07-10:
###8 July 25
I have been testing with two Raspberry Pi Zero W’s. I call them “wspr15m” and “wspr20m”. They have TAPR WSPR Hats and connect to different wifi networks. They both get loaded with 32 bit headless images but I have varied between Bookworm and Bullseye. 

I was successful getting wsprrypi to load when I set up the two pi zeros in my office…but they fail to connect to the network when outside of my office (such as in my kitchen or out in the yard near the antennas). In troubleshooting the wifi problem, I ended up rebuilding both pi zero w many times, uncovering the install script issue (ticket #209).

I shifted to building the images with the user pi and the password raspberry to try and reduce install script issues.

I also took to installing wsprry with the more complex “curl -sfSL installwspr.aa0nt.net | sudo bash -s -- debug”

###9 July 25
On your recommendation, I rebuild the (both non-working as WSPR) pi zero w’s.

**wspr15m**
removed TAPR hat. reimaged with 32bit bookwork lite. Powered up in the field and got wifi.
sudo apt update and upgraded. Started adding wsprrypi at 1336L. Locked up 1417L. the wsprrypi log is attached as wsprrypi-piwz15.log
rebooted and reinstalled wsprrypi. I expected the wsprrypi log to be overwritten but it show the 1336 start time data and then picks up at 1559L with the second attempt. That is the attached log wsprrypi-piwz15take2.log.  The wsprrypi install stopped responding and crashed the connection (ssh broken pipe) so I rebooted and reinstalled wsprrypi. That install completed but the wifi is not reliable. The log for the third install is attached as wsprrypi-15final.logfinal.log

[wsprrypi-piwz15.log](https://github.com/user-attachments/files/21152881/wsprrypi-piwz15.log)

[wsprrypi-piwz15take2.log](https://github.com/user-attachments/files/21152884/wsprrypi-piwz15take2.log)

[wsprrypi-15final.log](https://github.com/user-attachments/files/21152893/wsprrypi-15final.log)

I am back to troubleshooting the wifi on wspr15m.

**wspr20m**
reimaged with 32bit bullseye lite and then added wsprrypi in 1 build! …but wifi does not work. I stopped troubleshooting on wspr20m for now.

**Pi3A**
I imaged a raspberry pi 3A with 64 bit bookworm lite. wifi worked and I deployed it to the field and ssh’d in.
Via ssh, I sudo apt update and upgraded. via ssh, I installed wsprrypi. That log is attached as wsprrypi-3a.log. The ssh stopped (broken pipe) towards the end of the install. rebooted and reinstalled wsprrypi. That attempt is the attached log wsprrypi-pi15take1.log.

That install completed but the webpage was missing default values and would not save so I took your advice and reinstalled again. That install took and the install log for that run is attached as wsprrypi-3A15takefinal.log

I will continue to troubleshoot the wifi on the pi zeros.
I will install wsprrypi on a 4B and see if I can get that one in one install.
[wsprrypi-3A.log](https://github.com/user-attachments/files/21152925/wsprrypi-3A.log)

[wsprrypi-3A15take1.log](https://github.com/user-attachments/files/21152929/wsprrypi-3A15take1.log)

[wsprrypi-3A15takefinal.log](https://github.com/user-attachments/files/21152931/wsprrypi-3A15takefinal.log)

jdburke2k, 2025-07-10:
...but it came back, right? So I just need to be more patient?

lbussy, 2025-07-10:
After it disconnected, I took a call, and then when I came back, all was well.  So, it's "a thing to try" and see what happens.

jdburke2k, 2025-07-23:
No...sort of. I think this root user issue is fine now but installs keep killing my network and I/O.
I have tried multiple times (maybe 6 with different permutations...fresh image, then update/upgrade, then wifi dongle setup, then wspr; or image, update/upgrade, then wspr, then dongle; or image, wifi, wspr; or image, wspr; etc.)
It is not working still but I did learn some things:
1) I was using Warp on MacOS to ssh in...and would get broken pipe errors typically. I shifted to just using the keyboard locally and then happened to shift from Warp to Terminal and had much better luck with ssh. It seems that Warp has some issues that show up as a lost ssh connection after a bit.
2) stop me if you have heard this before...pi zero wifi is crap. I have gotten pretty good at setting up wifi your way with a dongle and the priority scripts. that helps life a lot.
3) lately (wspr 2.1.1) when I setup wspr on a well behaving pi zero w (either via ssh or with direct keyboard, monitor and ethernet) I get a syslog type dump on screen and no response from keyboard and then on reboot, I have no keyboard, no interfaces brought up (but I do get hdmi) and am locked out (so no log extraction). 

I think this is a new problem but I am installing on 2B now to see if it is just a pi zero thing or wider spread.

lbussy, 2025-07-23:
I have a Zero 2W where I confirmed it works. So Pi Zero 2W, Pi 3, and Pi 4, with a couple of people testing 2B for me.  Yes, WiFi is horrible.

There is an install log written to your user home; that would help determine what's happening.

jdburke2k, 2025-07-23:
I feel comfortable closing this ticket...the root user thing seems to be gone. I can not pull the install log on the Pi Zero W (not zero 2 w) because I am locked out...no ssh, no keyboard recognition. hdmi out works and I see it is assigned 127.0.0.1.

The 2Bs seem to take wsprry well and are up and on the air but I have not been able to confirm I am getting out (like I was before I went through all this rebuild on the pi zero w's). I think me not being heard is not a wsprry problem either.

I will do a little bit more troubleshooting with the pi zero and pi zero w but it seems to me that a better step might be to exclude it from the supported platforms (can we even buy pi zero anymore?).

lbussy, 2025-08-03:
Let me know what you find when you test more on the Pi Zero ... I don't have one and cannot buy one new, but it may be that we are asking too much out of the poor things.

closed 2025-08-03 by lbussy

jdburke2k, 2025-08-03:
I ended up being able to install wsprrypi on various raspberry pis, but failed to ever be heard by any stations through a TAPR filter board. I shifted to the older, wilfred image (WSPR_11_Feb_2019 2.img) and then both my 15m and 20m wsprs are heard and reported on wsprnet.org <http://wsprnet.org/>.

I will continue to troubleshoot the wsprrypi output on pi 2b or 3’s that I have and see if I can figure out more why wilfred works but wsprrypi does not.

Jd
AC0MM

lbussy, 2025-08-03:
Well, that's just weird .... I can definitely be heard when I test.  I decode locally to my receiver, and I also transmit through my OCFHW when not breaking things in between.

jdburke2k, 2025-08-06:
> lbussy
>  left a comment
> (lbussy/WsprryPi#209)
> Traveling till next weekend so I can't look at it. It's strange because I test by at least transmitting across the room. I have noted that you almost always need to put an attenuator on it, and maybe set TX power to 0, if you are doing that.

Thanks for the advice. I will turn the power down (something that is easy but didn’t think to try).