WsprryPi/WsprryPi #213: New Install on Raspberry Pi Zero 2 W Not Working

lbussy, 2025-07-11:
### Discussed in https://github.com/lbussy/WsprryPi/discussions/211

<div type='discussions-op-text'>

<sup>Originally posted by **goobius** July 11, 2025</sup>
Hello,

I am trying out WSPRRyPi on the RPi Zero 2 W and I can't get it is to transmit. 

This is the error I am receiving. Notice the Antenna Icon as it changes from Orange to Red. 

[image]

When I am SSH'ed into the RPi, this is what I am seeing for the status of the wsprrypi service: 

wsprrypi.service - Wsprry Pi daemon for: wsprrypi
     Loaded: loaded (/etc/systemd/system/wsprrypi.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Fri 2025-07-11 11:21:19 CDT; 1s ago
       Docs: https://github.com/lbussy/WsprryPi/discussions
    Process: 4008 ExecStart=/usr/local/bin/wsprrypi -D -i /usr/local/etc/wsprrypi.ini (code=killed, signal=SEGV)
   Main PID: 4008 (code=killed, signal=SEGV)
        CPU: 42ms

There is a killed process that I think is causing the issue. Has anyone seen this issue before?

Here are the logs for wsprrypi as well. I trying this out on 2m so I can hear it on my HT. 

2025-07-11 16:27:48.515 UTC     [ERROR] Try: wsprrypi --help
2025-07-11 16:27:53.722 UTC     [INFO ] WsprryPi version 2.1.1 (main).
2025-07-11 16:27:53.728 UTC     [INFO ] Running on: Raspberry Pi Zero 2 W Rev 1.0, 32-bit OS.
2025-07-11 16:27:53.728 UTC     [INFO ] Process PID: 5253
2025-07-11 16:27:53.728 UTC     [INFO ] Web server started on port: 31415
2025-07-11 16:27:53.729 UTC     [INFO ] WebSocketServer started on port: 31416
2025-07-11 16:27:53.730 UTC     [INFO ] WSPR loop running.
2025-07-11 16:27:53.731 UTC     [WARN ] Ignoring invalid frequency token: 144.489MHz
2025-07-11 16:27:53.731 UTC     [ERROR] Empty or invalid frequency list; disabling transmission.
2025-07-11 16:27:53.731 UTC     [FATAL] Missing required parameters.
2025-07-11 16:27:53.732 UTC     [ERROR] - At least one frequency must be specified.

Thanks for any help!
Keith
KI5DFU

</div>

lbussy, 2025-07-11:
@goobius, available bands are [specified here](https://github.com/lbussy/WsprryPi/blob/e61de5c490edd1e68304cab6ba44837a97b06ce0/src/wspr_band_lookup.cpp#L52-L81).

I'm still running around, but I did determine that 144.489MHz in the frequency field is crashing it.  Change it to `144489000` if you really want that frequency.

This is a bug, but I need to find a moment to fix it.

In the meantime:

`sudo nano /usr/local/etc/esprrypi.ini`

... and edit that line manually.

goobius, 2025-07-11:
Ok. I'm on my home network and still running into this issue:

[image]

I've googled the snot out of this process killer by USR1, but nothing is making sense. 

This is the error log for wsprrypi:
2025-07-11 23:25:03.467 UTC     [INFO ] WsprryPi version 2.1.1 (main).
2025-07-11 23:25:03.472 UTC     [INFO ] Running on: Raspberry Pi Zero 2 W Rev 1.                                                0, 64-bit OS.
2025-07-11 23:25:03.472 UTC     [INFO ] Process PID: 3712
2025-07-11 23:25:03.473 UTC     [INFO ] Web server started on port: 31415
2025-07-11 23:25:03.474 UTC     [INFO ] WebSocketServer started on port: 31416
2025-07-11 23:25:03.475 UTC     [INFO ] WSPR loop running.
2025-07-11 23:25:03.476 UTC     [WARN ] Ignoring invalid frequency token: 144.88                                                9MHz
2025-07-11 23:25:03.476 UTC     [ERROR] Empty or invalid frequency list; disabli                                                ng transmission.
2025-07-11 23:25:03.476 UTC     [FATAL] Missing required parameters.
2025-07-11 23:25:03.477 UTC     [ERROR] - At least one frequency must be specifi                                                ed.
2025-07-11 23:25:03.477 UTC     [ERROR] Try: wsprrypi --help

I did not use any VHF frequency. I input 20m as the band designation and still errors

lbussy, 2025-07-11:
This line:

> 2025-07-11 23:25:03.476 UTC [WARN ] Ignoring invalid frequency token: 144.88 9MHz

... shows it has not updated the INI.  Either edit `sudo nano /usr/local/etc/esprrypi.ini` manually, or delete it and reinstall and it will start over clean.

The program needs to run to allow a save from the UI to work.

lbussy, 2025-07-11:
I have this fixed in my current feature branch, so it will come out when I finish that work.  In the meantime, use a double representation.

commit 8372decf (2025-07-11): Fixes #213
PR #216 (2025-08-03): Fix for #213
Closes #213
commit 3e55ac11 (2025-08-03): Fix for #213 (#216)

* Closes #210

* Re-do WSPR-Transmitter callbacks

* Cleanup submodules on checkout or merge

* Fix callback logging

* Add Mailbox debug

* Closes #213

* Update to new FA package

* Remove audio vs blacklist