WsprryPi/WsprryPi #215: Signal timing broken in v2.1.1

GI3JMC, 2025-07-26:
Using Pi Zero 2w, Latest Raspberry OS 32-bit 'Lite', freshly imaged and updated.
Installed wsprrypi using the piped script and it completes without fault.
Configuration webpage accepts Callsign/Locator/Frequency/GPIO/etc and saves sucessfully.  After enabling transmit, the software waits until the next transmission window, then transmits for 900 seconds.  The signal is not a standard 2-minute wspr transmission repeated for 15 minutes, it appears to be 'stretched'.

The log showing 900+ second transmissions
[wsprrypi log.txt](https://github.com/user-attachments/files/21448778/wsprrypi.log.txt)

Waterfall Images showing the 'stretch' captured from a monitoring transceiver connected to wsjt-x
[image]
[image]

lbussy, 2025-07-26:
I'll have a look.  On my phone, what frequency?

GI3JMC, 2025-07-26:
Operating on 40m 7.0401MHz.  Changing the frequency between band name and 7040100Hz didn't make any difference.

lbussy, 2025-07-26:
Weird, that's one of the ones I use.  I'll try it later.

lbussy, 2025-07-27:
Whelp ... yup.  That's freakin' weird.

Let me give it a look-see.

lbussy, 2025-08-01:
Okay, I'm home.  I reproduced the issue quickly. Thank you for trying out that 32-bit vs 64-bit outcome.

Working on it.

lbussy, 2025-08-02:
In the meantime, anyone seeing this should use a 64-bit version of Raspbian.

lbussy, 2025-08-06:
I did not get it figured out before I took off again for another trip. I'll be gone for about a week and a half. I did also confirm that 2200 m has the same incorrect output.

lbussy, 2025-09-04:
Still here ... not set up my bench yet.

lbussy, 2026-01-18:
I am still working on this - the problem is it's tied to a major shift in my code, so it's not going to be quick (especially with my day job killing me).